coolsnowwolf/lede #8739: npc

nanbowan123, 2022-01-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.285 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
### (1) 具体问题

A：Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.285 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout

### (2) 路由器型号和固件版本

A：X86 固件版本OpenWrt R22.1.1 / LuCI Master (git-21.335.48743-5f363d9)

### (3) 详细日志

A：Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.174 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.174 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.174 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.174 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.174 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.175 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.175 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.175 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.175 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.175 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.176 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.176 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.176 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.176 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.274 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.275 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.275 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.276 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.276 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.276 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.276 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.276 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.276 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.276 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.276 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.276 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.276 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.276 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout
Thu Jan 20 22:42:19 2022 daemon.info npc[7655]: 2022/01/20 14:42:19.285 [1;33m[W][0m [client.go:213]  connect to www.xn11xn.com:80 error dial tcp 172.67.37.102:80: i/o timeout

Her0R0cket, 2022-01-21:
你这npc连的服务器是正确的吗,确定是你的吗?这网址少儿不宜啊.

WYC-2020, 2022-01-21:
看破不说破哈哈