coolsnowwolf/lede #8740: 5.10内核NAT环回似乎失效问题

hamlinzheng, 2022-01-20:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

使用本Repo Actions编译的固件，配置了端口转发并开启`启用NAT环回`，外网可使用域名+端口正常访问，内网使用域名+端口无法访问，参考 #6149 似乎配置方式已经不一样，暂未找到解决办法，寻求帮助~

### (2) 路由器型号和固件版本

PVE+Lede
内核版本：5.10.92

```
$ cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_TAINTS='no-all'
DISTRIB_REVISION='R22.1.1'
DISTRIB_DESCRIPTION='OpenWrt '
```

hamlinzheng, 2022-01-20:
补充：
- `/etc/sysctl.d`目录下没有`br-netfilter`相关配置文件
- 在`/etc/sysctl.conf`中加入下列配置并执行`sysctl -p`后会出现报错

```
## /etc/sysctl.conf新增配置
net.bridge.bridge-nf-call-arptables=0
net.bridge.bridge-nf-call-ip6tables=0
net.bridge.bridge-nf-call-iptables=0

## 报错
sysctl: error: 'net.bridge/bridge-nf-call-arptables' is an unknown key
sysctl: error: 'net.bridge/bridge-nf-call-ip6tables' is an unknown key
sysctl: error: 'net.bridge/bridge-nf-call-iptables' is an unknown key

```

alexyw, 2022-01-23:
你的/etc/sysctl.d下面都有啥？可以逐个打开看看，有个文件里面肯定写了

hamlinzheng, 2022-01-23:
> 你的/etc/sysctl.d下面都有啥？可以逐个打开看看，有个文件里面肯定写了

逐个打开看过，没有`net.bridge/bridge-nf-call-xxx`相关配置，已经准备刷旧内核的固件了

```
root@OpenWrt:~# ls /etc/sysctl.d/
10-default.conf       11-nf-conntrack.conf  12-tcp-bbr.conf

```

WYC-2020, 2022-01-24:
我这边有是在5.15上面测试的
[image]
[image]

hamlinzheng, 2022-01-24:
> 我这边有是在5.15上面测试的 [image] [image]

这是自己编译的吗？唉，暂时没时间研究自编译固件==

WYC-2020, 2022-01-24:
嗯自己编译的 @hamlinzheng

socekin, 2022-05-03:
> 我这边有是在5.15上面测试的 [image] [image]

10-default.conf           11-nf-conntrack.conf      12-tcp-bbr.conf           sysctl-net-core-mem.conf

我自己按源码编译只有这几个文件，没有11-br

steamch, 2022-06-29:
我也遇到了同样的问题，自己编译的
`## 报错
sysctl: error: 'net.bridge/bridge-nf-call-arptables' is an unknown key
sysctl: error: 'net.bridge/bridge-nf-call-ip6tables' is an unknown key
sysctl: error: 'net.bridge/bridge-nf-call-iptables' is an unknown key`

coolsnowwolf, 2026-09-05:
NAT loopback 报告基于 2022 年 Linux 5.10 和旧版防火墙配置，没有当前 firewall4/nftables 配置、规则计数或复现日志；无法确认现行代码缺陷，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf