coolsnowwolf/lede #8741: 5.15内核编译出来显示dnsmasq启动失败

aakkll, 2022-01-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：没办法正常上网，但是开了ssrp后，外网可以的，国内网是不行的

### (2) 路由器型号和固件版本

A：X86_64

### (3) 详细日志

A：
Fri Jan 21 01:14:11 2022 daemon.err nlbwmon[3818]: The netlink receive buffer size of 1048576 bytes will be capped to 212992 bytes
Fri Jan 21 01:14:11 2022 daemon.err nlbwmon[3818]: by the kernel. The net.core.rmem_max sysctl limit needs to be raised to
Fri Jan 21 01:14:11 2022 daemon.err nlbwmon[3818]: at least 1048576 in order to sucessfully set the desired receive buffer size!
Fri Jan 21 01:14:11 2022 daemon.notice procd: /etc/rc.d/S90kms: KMS Server has started.
Fri Jan 21 01:14:11 2022 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Fri Jan 21 01:14:12 2022 daemon.crit dnsmasq[4240]: bad SRV target at line 29 of /var/etc/dnsmasq.conf.cfg01411c
Fri Jan 21 01:14:12 2022 daemon.crit dnsmasq[4240]: FAILED to start up
Fri Jan 21 01:14:12 2022 daemon.notice procd: /etc/rc.d/S90turboacc: net.ipv4.tcp_congestion_control = cubic
Fri Jan 21 01:14:12 2022 daemon.notice procd: /etc/rc.d/S90turboacc: killall: dnscache: no process killed
Fri Jan 21 01:14:12 2022 daemon.notice procd: /etc/rc.d/S90turboacc: sh: you need to specify whom to kill
Fri Jan 21 01:14:12 2022 daemon.notice procd: /etc/rc.d/S90turboacc: Stop DNS Caching
Fri Jan 21 01:14:12 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of lan (br-lan)
Fri Jan 21 01:14:12 2022 daemon.notice procd: /etc/rc.d/S90turboacc: DNSMASQ change
Fri Jan 21 01:14:12 2022 daemon.crit dnsmasq[4611]: bad SRV target at line 29 of /var/etc/dnsmasq.conf.cfg01411c
Fri Jan 21 01:14:12 2022 daemon.crit dnsmasq[4611]: FAILED to start up
Fri Jan 21 01:14:12 2022 daemon.notice procd: /etc/rc.d/S91wrtbwmon: Warning: using existing /tmp/usage.db
Fri Jan 21 01:14:12 2022 authpriv.info dropbear[4934]: Not backgrounding
Fri Jan 21 01:14:12 2022 daemon.crit dnsmasq[5497]: bad SRV target at line 29 of /var/etc/dnsmasq.conf.cfg01411c
Fri Jan 21 01:14:12 2022 daemon.crit dnsmasq[5497]: FAILED to start up
Fri Jan 21 01:14:12 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of loopback (lo)
Fri Jan 21 01:14:13 2022 daemon.crit dnsmasq[5943]: bad SRV target at line 29 of /var/etc/dnsmasq.conf.cfg01411c
Fri Jan 21 01:14:13 2022 daemon.crit dnsmasq[5943]: FAILED to start up
Fri Jan 21 01:14:14 2022 user.info mwan3rtmon[4014]: Detect rtchange event.
Fri Jan 21 01:14:18 2022 daemon.crit dnsmasq[6072]: bad SRV target at line 29 of /var/etc/dnsmasq.conf.cfg01411c
Fri Jan 21 01:14:18 2022 daemon.crit dnsmasq[6072]: FAILED to start up
Fri Jan 21 01:14:21 2022 daemon.err uhttpd[3756]: cat: write error: Broken pipe

WYC-2020, 2022-01-20:
我这边一切正常呢

coolsnowwolf, 2022-01-21:
bad SRV target at line 29 of /var/etc/dnsmasq.conf.cfg01411c

aakkll, 2022-01-21:
> bad SRV target at line 29 of /var/etc/dnsmasq.conf.cfg01411c

我知道什么问题了

closed 2022-01-21 by aakkll