coolsnowwolf/lede #8747: 编译5.4和5.10内核的固件报ntfs3的错误

aakkll, 2022-01-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译报错

### (2) 路由器型号和固件版本

A：X86_64

### (3) 详细日志

A：
Configuring ddns-scripts_freedns_42_pl.
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kmod-fs-ntfs3 for ntfs3-mount
 * pkg_hash_fetch_best_installation_candidate: Packages for ntfs3-mount found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for automount:
 *      kmod-fs-ntfs3
 * opkg_install_cmd: Cannot install package automount.
 * check_data_file_clashes: Package antfs-mount wants to install file /home/leung/kk15/lede/build_dir/target-x86_64_musl/root-x86/sbin/mount.ntfs
        But that file is already provided by package  * ntfs3-mount
 * opkg_install_cmd: Cannot install package antfs-mount.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/leung/kk15/lede'
make[1]: *** [package/Makefile:111: /home/leung/kk15/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/leung/kk15/lede'
make: *** [/home/leung/kk15/lede/include/toplevel.mk:230：world] 错误 2

Cptmacmillan2022007, 2022-01-21:
pkg hash值不对要更新

WYC-2020, 2022-01-22:
去掉编译ntfs3-mount这是5.15内核的，如果你是用的lede原代码去掉编译这个或者删除config重新配置应该就没问题，如果不是就看下那个automount的依赖哪里是否限制了内核版本

closed 2022-01-24 by aakkll