coolsnowwolf/lede #8751: 条件编译的问题

HiGarfield, 2022-01-21:
https://github.com/coolsnowwolf/lede/blob/95f5440b13e0a3a18c3de2bf3b59dd4479b83168/target/linux/generic/hack-5.4/953-net-patch-linux-kernel-to-support-shortcut-fe.patch#L179
感觉这个条件编译是多于的。

看到
https://github.com/coolsnowwolf/lede/blob/95f5440b13e0a3a18c3de2bf3b59dd4479b83168/target/linux/generic/hack-5.4/952-net-conntrack-events-support-multiple-registrant.patch#L144
这里面，如果CONFIG_NF_CONNTRACK_CHAIN_EVENTS宏定义存在，那么整个nf_conntrack_eventmask_report函数都是用的952补丁中重新定义过的版本，根本不会编译到原始版本里面来，在原始版本版本中加CONFIG_NF_CONNTRACK_CHAIN_EVENTS的条件编译是否没有意义？

@AmadeusGhost

WYC-2020, 2022-01-22:
确实没必要了，可以去掉，无所谓的

WYC-2020, 2022-01-22:
怀疑这两个补丁不是在前一个补丁的基础上做的，所以会有这种缺陷

coolsnowwolf, 2026-09-06:
该条件编译疑问针对旧 5.4 shortcut-fe 补丁，当前内核和补丁栈已演进，没有现行编译失败或行为问题，关闭为“不计划处理”。

closed 2026-09-06 by coolsnowwolf