coolsnowwolf/lede #8753: 编译报错：缺少依赖库 hwmon.ko

maxelblack, 2022-01-22:
## 1.关于提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译 Linux 内核时报错找不到依赖库 hwmon.ko

"Package kmod-bnxt-en is missing dependencies for the following libraries: hwmon.ko"

### (2) 路由器型号和固件版本

A：Xiaomi Mi Router 3 Pro

### (3) 详细日志

A：
```
make[3]: Entering directory '/home/compile/coolsnowwolf/lede/package/kernel/linux'
mkdir -p /home/compile/coolsnowwolf/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp

SHELL= flock /home/compile/coolsnowwolf/lede/tmp/.root-copy.flock -c 'cp -fpR /home/compile/coolsnowwolf/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/packages/.pkgdir/kernel/. /home/compile/coolsnowwolf/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/compile/coolsnowwolf/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.kernel_installed
Package kmod-bnxt-en is missing dependencies for the following libraries:
hwmon.ko
make[3]: *** [modules/netdevices.mk:1141: /home/compile/coolsnowwolf/lede/bin/targets/ramips/mt7621/packages/kmod-bnxt-en_5.4.172-1_mipsel_24kc.ipk] Error 1
make[3]: Leaving directory '/home/compile/coolsnowwolf/lede/package/kernel/linux'
time: package/kernel/linux/compile#3.44#0.40#3.81
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/compile/coolsnowwolf/lede'
make[1]: *** [package/Makefile:110: /home/compile/coolsnowwolf/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/compile/coolsnowwolf/lede'
make: *** [/home/compile/coolsnowwolf/lede/include/toplevel.mk:230: world] Error 2
```
由于是在 screen 环境里运行的，只能找到这一点日志了（screen不支持滚动）

WYC-2020, 2022-01-22:
缺钱依赖而已

WYC-2020, 2022-01-22:
@maxelblack 加上这个看下
[image]

maxelblack, 2022-01-23:
好，有空试试

maxelblack, 2022-01-24:
> @maxelblack 加上这个看下 [image]

hwmon.ko 的依赖解决了，但又出新的问题了：

```
Package kmod-multimedia-input is missing dependencies for the following libraries:
input-core.ko
make[3]: *** [modules/video.mk:238: /home/compile/coolsnowwolf/lede/bin/targets/ramips/mt7621/packages/kmod-multimedia-input_5.4.172-1_mipsel_24kc.ipk] Error 1
make[3]: Leaving directory '/home/compile/coolsnowwolf/lede/package/kernel/linux'
time: package/kernel/linux/compile#43.32#31.43#97.60
    ERROR: package/kernel/linux failed to build.
```

是我编译环境的问题吗（用的是 Rocky Linux 8.5 ）

WYC-2020, 2022-01-24:
不好意思我截图接错了，是在
[image]
这里加 ，如果还不行到时候我编译看下 ，你先试下@maxelblack

maxelblack, 2022-01-24:
> 不好意思我截图接错了，是在 [image] 这里加 ，如果还不行到时候我编译看下 ，你先试下@maxelblack

就是在这里加的，然而这个好了另一个依赖却出问题了

WYC-2020, 2022-01-24:
下午看下

WYC-2020, 2022-01-24:
依赖这些其实非常简单

WYC-2020, 2022-01-24:
@maxelblack 
[image]

maxelblack, 2022-01-24:
谢啦，我去试试

closed 2022-02-28 by AmadeusGhost