coolsnowwolf/lede #8754: 编译报错

Karmylr, 2022-01-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
[image]

### (2) 路由器型号和固件版本

A：j4125

### (3) 详细日志

A：
tar: Exiting with failure status due to previous errors
make[4]: *** [Makefile:29: /home/conter/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.93/.prepared_ce10e32be049c2d006f40939287b23e4] Error 2
make[4]: Leaving directory '/home/conter/lede/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/conter/lede/target/linux'
time: target/linux/compile#1.59#13.01#19.91
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/conter/lede'
make[1]: *** [target/Makefile:18: /home/conter/lede/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/conter/lede'
make: *** [/home/conter/lede/include/toplevel.mk:230: world] Error 2

AmadeusGhost, 2022-01-22:
你硬盘没有空间了。

closed 2022-01-22 by AmadeusGhost

Karmylr, 2022-01-22:
> 你硬盘没有空间了。

太谢谢了