coolsnowwolf/lede #8755: 编译报错

matianhe3, 2022-01-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
ubuntu 20.04

### (3) 详细日志

```
Applying /home/lean/lede/target/linux/bcm27xx/patches-5.10/950-0733-usb-xhci-workaround-for-bogus-SET_DEQ_PENDING-endpoi.patch using plaintext: 
patching file drivers/usb/host/xhci-ring.c
Hunk #1 succeeded at 4268 (offset 10 lines).

Applying /home/lean/lede/target/linux/bcm27xx/patches-5.10/950-0734-overlays-Add-TI-BQ32000-RTC-support.patch using plaintext: 
patching file arch/arm/boot/dts/overlays/README
patching file arch/arm/boot/dts/overlays/i2c-rtc-common.dtsi

Applying /home/lean/lede/target/linux/bcm27xx/patches-5.10/950-0735-xhci-guard-accesses-to-ep_state-in-xhci_endpoint_res.patch using plaintext: 
patching file drivers/usb/host/xhci.c
Hunk #1 FAILED at 3270.
Hunk #2 FAILED at 3359.
2 out of 2 hunks FAILED -- saving rejects to file drivers/usb/host/xhci.c.rej
Patch failed!  Please fix /home/lean/lede/target/linux/bcm27xx/patches-5.10/950-0735-xhci-guard-accesses-to-ep_state-in-xhci_endpoint_res.patch!
make[4]: *** [Makefile:44: /home/lean/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/linux-5.10.93/.prepared_697c268227bc712577f34e0c83a4cbc7] Error 1
make[4]: Leaving directory '/home/lean/lede/target/linux/bcm27xx'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/lean/lede/target/linux'
time: target/linux/compile#1.95#10.94#9.25
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/lean/lede'
make[1]: *** [target/Makefile:18: /home/lean/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/lean/lede'
make: *** [/home/lean/lede/include/toplevel.mk:230: world] Error 2
```

WYC-2020, 2022-01-22:
更新patch文件，这个自己修复一下就可以了

Beginner-Go, 2022-01-22:
我对比了下上游 把这个patch删除就好了

Beginner-Go, 2022-01-22:
测试没问题反馈一下

matianhe3, 2022-01-22:
> 测试没问题反馈一下

结束还是有error 但是镜像编译出来了， 也可以正常使用。

closed 2022-01-22 by matianhe3

Beginner-Go, 2022-01-22:
什么error

matianhe3, 2022-01-22:
> 什么error
我看命令行里输出 img 文件编译出来了， 我就去bin目录找了，  没有把error复制出来。