coolsnowwolf/lede #8757: 编译错误，缺少依赖？

moqingzhu1, 2022-01-22:
## 1.关于你要提交的问题

## 2. 详细叙述
只要带SSR就会编译失败，重新GIT也没用

### (1) 具体问题

A：缺少依赖

### (2) 路由器型号和固件版本

A：星际宝盒

### (3) 详细日志

A：如下固件，星际宝盒

  Target System (Qualcomm Atheros IPQ40XX)  --->               │ │  

│ │ Target Profile (MobiPromo CM520-79F) --->

全局科学上网

Configuring xray-core.
Collected errors:

    pkg_hash_check_unresolved: cannot find dependency batctl-default for luci-app-easymesh
    pkg_hash_check_unresolved: cannot find dependency kmod-batman-adv for luci-app-easymesh
    pkg_hash_fetch_best_installation_candidate: Packages for luci-app-easymesh found, but incompatible with the architectures configured
    opkg_install_cmd: Cannot install package luci-app-easymesh.
    satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-easymesh-zh-cn:
    batctl-default
    kmod-batman-adv
    opkg_install_cmd: Cannot install package luci-i18n-easymesh-zh-cn.
    pkg_hash_check_unresolved: cannot find dependency shadowsocksr-libev-ssr-redir for luci-app-vssr
    pkg_hash_check_unresolved: cannot find dependency shadowsocksr-libev-ssr-check for luci-app-vssr
    pkg_hash_check_unresolved: cannot find dependency shadowsocksr-libev-ssr-server for luci-app-vssr
    pkg_hash_fetch_best_installation_candidate: Packages for luci-app-vssr found, but incompatible with the architectures configured
    opkg_install_cmd: Cannot install package luci-app-vssr.
    satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-vssr-zh-cn:
    shadowsocksr-libev-ssr-redir
    shadowsocksr-libev-ssr-check
    shadowsocksr-libev-ssr-server
    opkg_install_cmd: Cannot install package luci-i18n-vssr-zh-cn.
    package/Makefile:66: recipe for target 'package/install' failed
    make[2]: *** [package/install] Error 255
    make[2]: Leaving directory '/home/juedy/lede'
    package/Makefile:111: recipe for target '/home/juedy/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_install' failed
    make[1]: *** [/home/juedy/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_install] Error 2
    make[1]: Leaving directory '/home/juedy/lede'

mitsukileung, 2022-01-22:
[VSSR上面有说明](https://github.com/jerrykuku/luci-app-vssr)

使用lede最新源码编译失败，报错缺少依赖：

satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-vssr:
- shadowsocksr-libev-ssr-local
- shadowsocksr-libev-ssr-redir
- shadowsocksr-libev-ssr-check
- xray-core
- xray-plugin
- shadowsocksr-libev-ssr-server
opkg_install_cmd: Cannot install package luci-app-vssr.
原因是lede缺少软件源，解决办法，清除缓存重新下载编译：

# 1.清除缓存
rm -rf tmp

# 2.feeds.conf文件添加源
src-git helloworld https://github.com/fw876/helloworld
src-git passwall https://github.com/xiaorouji/openwrt-passwall

# 3.重新执行升级安装下载编译等操作
./scripts/feeds update -a
./scripts/feeds install -a
make -j8 download V=s
make -j1 V=s

moqingzhu1, 2022-01-22:
按照VSSR上面一字不差复制的都不行

WYC-2020, 2022-01-22:
你怕是找错地方了，谁的vssr就去找谁

moqingzhu1, 2022-01-22:
> 你怕是找错地方了，谁的vssr就去找谁
不是说VSSR有问题，PASSWALL，HELLOWORD，SSRPLUS，ESAYMESH都这样

WYC-2020, 2022-01-22:
我这边没任何问题

moqingzhu1, 2022-01-27:
问题解决，下载的源码不完整

closed 2022-02-12 by Beginner-Go