coolsnowwolf/lede #8758: 强迫症一个时间显示小建议

tianiue, 2022-01-22:
[image]
建议换一种显示方式，改成中文或者纯数字方式
2022年1月22日 星期六 12:00:06
2022/1/22 周六 12:00:06

Beginner-Go, 2022-01-22:
可以 但没必要

hzokdj, 2022-01-23:
好提议，期待

deplives, 2022-01-23:
你可以自行修改对应架构的index，比如x86的是
https://github.com/coolsnowwolf/lede/blob/744d1ed2c7d79b3554ead609716ebaab6f1ed350/package/lean/autocore/files/x86/index.htm#L63
比如将这一行改为
`localtime  = os.date("%Y-%m-%d %H:%M:%S")`
则为 `2022-01-23 12:45:46` 这种时间格式

GreatMichaelLee, 2022-01-23:
> 你可以自行修改对应架构的index，比如x86的是
> 
> https://github.com/coolsnowwolf/lede/blob/744d1ed2c7d79b3554ead609716ebaab6f1ed350/package/lean/autocore/files/x86/index.htm#L63
> 
> 
> 比如将这一行改为
> `localtime = os.date("%Y-%m-%d %H:%M:%S")`
> 则为 `2022-01-23 12:45:46` 这种时间格式

这个建议比上面的有营养

tianiue, 2022-01-24:
> 你可以自行修改对应架构的index，比如x86的是
> 
> https://github.com/coolsnowwolf/lede/blob/744d1ed2c7d79b3554ead609716ebaab6f1ed350/package/lean/autocore/files/x86/index.htm#L63
> 
> 
> 比如将这一行改为
> `localtime = os.date("%Y-%m-%d %H:%M:%S")`
> 则为 `2022-01-23 12:45:46` 这种时间格式

请问云编译不修改源码如何添加替换？

deplives, 2022-01-24:
> tianiue

在你自己的脚本里面sed替换

deplives, 2022-01-24:
> > 你可以自行修改对应架构的index，比如x86的是
> > https://github.com/coolsnowwolf/lede/blob/744d1ed2c7d79b3554ead609716ebaab6f1ed350/package/lean/autocore/files/x86/index.htm#L63
> > 
> > 比如将这一行改为
> > `localtime = os.date("%Y-%m-%d %H:%M:%S")`
> > 则为 `2022-01-23 12:45:46` 这种时间格式
> 
> 请问云编译不修改源码如何添加替换？

在你自己的脚本里面用 sed 替换

tianiue, 2022-01-24:
> > > 你可以自行修改对应架构的index，比如x86的是
> > > https://github.com/coolsnowwolf/lede/blob/744d1ed2c7d79b3554ead609716ebaab6f1ed350/package/lean/autocore/files/x86/index.htm#L63
> > > 
> > > 比如将这一行改为
> > > `localtime = os.date("%Y-%m-%d %H:%M:%S")`
> > > 则为 `2022-01-23 12:45:46` 这种时间格式
> > 
> > 
> > 请问云编译不修改源码如何添加替换？
> 
> 在你自己的脚本里面用 sed 替换

2022年1月22日 星期六 12:00:06 这个格式不会写

deplives, 2022-01-24:
> 2022年1月22日 星期六 12:00:06 这个格式不会写

你这个啥都要喂到嘴里么，学会用一下搜索引擎

closed 2022-01-24 by tianiue