coolsnowwolf/lede #8761: 大佬们编译k3固件的时候会额外生成其它bcm53xx机型的固件？

weigeloveu, 2022-01-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
编译k3固件的时候会额外生成其它bcm53xx机型的固件,配置文件已经上传，是我哪里没设置好吗？

### (2) 路由器型号和固件版本

A：k3

### (3) 详细日志

A：```
CONFIG_MODULES=y
CONFIG_HAVE_DOT_CONFIG=y
# CONFIG_TARGET_sunxi is not set
# CONFIG_TARGET_apm821xx is not set
# CONFIG_TARGET_ath25 is not set
# CONFIG_TARGET_ath79 is not set
# CONFIG_TARGET_bcm27xx is not set
CONFIG_TARGET_bcm53xx=y
# CONFIG_TARGET_bcm47xx is not set
# CONFIG_TARGET_bcm4908 is not set
# CONFIG_TARGET_bcm63xx is not set
# CONFIG_TARGET_bmips is not set
# CONFIG_TARGET_octeon is not set
# CONFIG_TARGET_gemini is not set
# CONFIG_TARGET_mpc85xx is not set
# CONFIG_TARGET_mxs is not set
# CONFIG_TARGET_lantiq is not set
# CONFIG_TARGET_malta is not set
# CONFIG_TARGET_pistachio is not set
# CONFIG_TARGET_mvebu is not set
# CONFIG_TARGET_kirkwood is not set
# CONFIG_TARGET_mediatek is not set
# CONFIG_TARGET_ramips is not set
# CONFIG_TARGET_at91 is not set
# CONFIG_TARGET_tegra is not set
# CONFIG_TARGET_layerscape is not set
# CONFIG_TARGET_imx is not set
# CONFIG_TARGET_octeontx is not set
# CONFIG_TARGET_oxnas is not set
# CONFIG_TARGET_armvirt is not set
# CONFIG_TARGET_ipq40xx is not set
# CONFIG_TARGET_ipq60xx is not set
# CONFIG_TARGET_ipq806x is not set
# CONFIG_TARGET_ipq807x is not set
# CONFIG_TARGET_realtek is not set
# CONFIG_TARGET_rockchip is not set
# CONFIG_TARGET_arc770 is not set
# CONFIG_TARGET_archs38 is not set
# CONFIG_TARGET_omap is not set
# CONFIG_TARGET_uml is not set
# CONFIG_TARGET_zynq is not set
# CONFIG_TARGET_x86 is not set
CONFIG_TARGET_bcm53xx_generic=y
# CONFIG_TARGET_MULTI_PROFILE is not set
# CONFIG_TARGET_bcm53xx_generic_Generic is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_asus_rt-ac56u is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_asus_rt-ac68u is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_asus_rt-ac87u is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_asus_rt-n18u is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_buffalo_wxr-1900dhp is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_buffalo_wzr-600dhp2 is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_buffalo_wzr-900dhp is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_buffalo_wzr-1750dhp is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_dlink_dir-885l is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_linksys_ea6500-v2 is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_linksys_ea9500 is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_luxul_abr-4500 is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_luxul_xap-1610 is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_luxul_xbr-4500 is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_luxul_xwr-3150 is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_meraki_mr32 is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_netgear_r6250 is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_netgear_r6300-v2 is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_netgear_r7000 is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_netgear_r7900 is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_netgear_r8000 is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_netgear_r8500 is not set
# CONFIG_TARGET_bcm53xx_generic_DEVICE_smartrg_sr400ac is not set
CONFIG_TARGET_bcm53xx_generic_DEVICE_phicomm_k3=y
```

destiny-dream, 2022-01-23:
会

weigeloveu, 2022-01-23:
> 会

大佬有什么解决办法吗？

destiny-dream, 2022-01-24:
> > 会
> 
> 大佬有什么解决办法吗？

记得在别的issues里好像有提过，具体忘了。我是不在意这个问题

dhxh, 2022-01-24:
sed -i 's|^TARGET_|# TARGET_|g; s|# TARGET_DEVICES += phicomm_k3|TARGET_DEVICES += phicomm_k3|' target/linux/bcm53xx/image/Makefile

dhxh, 2022-01-24:
把除k3以外的机型注释就可以了

weigeloveu, 2022-01-25:
> 把除k3以外的机型注释就可以了

多谢大佬，解决了

closed 2022-01-25 by weigeloveu