coolsnowwolf/lede #8762: dnsmasq占用率高且无法上网

zopargr, 2022-01-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：运行一段时间后，dnsmasq占用率特别高且无法上网，手动kill掉dnsmasq的进程，dnsmasq会自动重启，占用率依然居高不下，只能重启整个路由器解决，然后过了一天又会复现这个问题

### (2) 路由器型号和固件版本

A：X86 固件5.15.16

### (3) 详细日志

A：
[image]
[image]
[image]

zhangguanzhang, 2022-01-23:
👀 你需要 taskset 绑定限制核心

jlzsss, 2022-02-02:
好像还是没有修复？

zopargr, 2022-02-03:
卸载掉socat后恢复正常了，此issue关闭。

closed 2022-02-03 by zopargr