coolsnowwolf/lede #8763: R2S 自元旦后的版本LAN口就起不来了

GreatMichaelLee, 2022-01-23:
rt, R2S 自元旦后的基线就LAN口起不来，元旦之前12.31的版本都是好的，也不知道是什么原因，谁能给点提示该怎么排除....这种东西没日志，也不知道该怎么弄. partition scheme也没改。

mingxiaoyu, 2022-01-23:
patches-5.4/600-net-phy-Add-driver-for-Motorcomm-YT85xx-PHYs.patch
回退到这个

https://github.com/coolsnowwolf/lede/blob/b1b36c86af6d9fa3ba57eb2140025d2039996b81/target/linux/rockchip/patches-5.4/600-net-phy-Add-driver-for-Motorcomm-YT85xx-PHYs.patch

如果你可以帮我个小忙的话
把
https://github.com/mingxiaoyu/immortalwrt/blob/r1lts/target/linux/rockchip/patches-5.10/601-support-for-YT8531-S.patch
也加进去试试

这个补丁会影响 R2S不？

@GreatMichaelLee

GreatMichaelLee, 2022-01-23:
谢谢，我试试

GreatMichaelLee, 2022-01-23:
by the way, patches-5.4这个目录下的补丁是for 5.4内核的么，貌似我元旦前和元旦后都编译的是5.10内核的，确定跟这个相关吗？因为我设备在远端，有可能升挂....

mingxiaoyu, 2022-01-23:
哈哈。我只是知道r2c 被这个补丁能裂了

mingxiaoyu, 2022-01-23:
> 

5.10  那就应该没关系

467815891a, 2022-01-23:
我是删除.config，重新make menuconfig就可以了

GreatMichaelLee, 2022-01-24:
> 我是删除.config，重新make menuconfig就可以了

你也是遇到我这个同样的问题，重新renew .config修复的吗？

467815891a, 2022-01-24:
> > 我是删除.config，重新make menuconfig就可以了
> 
> 你也是遇到我这个同样的问题，重新renew .config修复的吗？

是的，`rm .config && make menuconfig`

skaterjuice, 2022-01-24:
How to add patch? What is the command?

closed 2022-01-25 by AmadeusGhost