coolsnowwolf/lede #8766: ssh连接其他设备异常Segmentation fault

ITdesk01, 2022-01-23:
### 路由器型号和固件版本
X86固件
OpenWrt R22.1.1 / LuCI Master (git-21.335.48743-5f363d9)
2022-01-23 18:33 (commit:95f5440b1)

 ### 具体问题
我尝试在openwrt里连接我的ubuntu,结果输完密码以后出现了Segmentation fault，我排查过ubuntu的问题，ubuntu用电脑端的ssh软件可以正常连接，我也尝试连接我的群辉，但问题依旧，每次都是输完密码然后直接Segmentation fault

具体命令如下
```
root@OpenWrt:~# ssh zhang@192.168.10.199
zhang@192.168.10.199's password:
Segmentation fault
```

hxlls, 2022-01-23:
[image]
实测不存在问题，先自己检查下把

xdhgsq, 2022-01-24:
> [image] 实测不存在问题，先自己检查下把

我又试了一次重新全编译，正常了，看起来应该是我长时间编译，有异常了

ITdesk01, 2022-01-24:
> [image] 实测不存在问题，先自己检查下把

正常了，编译太多次，源码抽风了