coolsnowwolf/lede #8768: 少了Newifi D2型号

sinian-liu, 2022-01-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
Target System选择 "MediaTek Ralink MIPS"  Subtarget选择"MT7621" 但Target Profile选择里面没有"Newifi D2"选项 能否重新添加该型号？
### (1) 具体问题

A：Target System选择 "MediaTek Ralink MIPS"  Subtarget选择"MT7621"  但Target Profile选择里面没有"Newifi D2"选项，试过很多次都找不到，是不是取消了该型号的支持？能否重新添加该型号？

### (2) 路由器型号和固件版本

A：Newifi D2

### (3) 详细日志

A：找不到对应的Newifi D2型号

EnnawYang, 2022-01-24:
？
[image]

sinian-liu, 2022-01-24:
> ？ [image]

thank you

closed 2022-01-25 by AmadeusGhost