coolsnowwolf/lede #8773: 现在x86的系统日志 不输出accepted login 之类的web登录日志了吗

82kg, 2022-01-25:
没有类似issues

### (1) 具体问题
不输出accepted login 之类的web登录日志，也没有ssh登录的日志

### (2) 路由器型号和固件版本
X86 
5.10.89内核

Wilson-coder, 2022-01-27:
```
Thu Jan 27 11:30:33 2022 daemon.warn miniupnpd[8095]: HTTP Connection from 192.168.0.214 closed unexpectedly
Thu Jan 27 11:30:33 2022 daemon.warn miniupnpd[8095]: HTTP Connection from 192.168.0.214 closed unexpectedly
Thu Jan 27 11:35:09 2022 authpriv.info dropbear[3102]: Child connection from 192.168.0.122:10798
Thu Jan 27 11:35:09 2022 authpriv.notice dropbear[3102]: Password auth succeeded for 'root' from 192.168.0.122:10798
Thu Jan 27 11:35:22 2022 authpriv.info dropbear[3102]: Exit (root) from <192.168.0.122:10798>: Exited normally
```

刚试了一下，很正常。内核5.10.93

82kg, 2022-01-27:
> ```
> Thu Jan 27 11:30:33 2022 daemon.warn miniupnpd[8095]: HTTP Connection from 192.168.0.214 closed unexpectedly
> Thu Jan 27 11:30:33 2022 daemon.warn miniupnpd[8095]: HTTP Connection from [...]

刚编译的吗？

Wilson-coder, 2022-01-27:
> 刚编译的吗？

最近的版本都没问题，10.92的时候刚好也看过系统日志，都有记录。

82kg, 2022-01-27:
> 最近的版本都没问题，10.92的时候刚好也看过系统日志，都有记录。

刚最新版试了，ssh登录，确实有记录的，但是web登录  好像没有。

coolsnowwolf, 2026-09-06:
Web/SSH 登录日志由 uhttpd、rpcd、dropbear/openssh 和 syslog 配置共同决定，报告没有当前服务配置和日志级别，关闭为“不计划处理”。

closed 2026-09-06 by coolsnowwolf