coolsnowwolf/lede #8774: R2S编译一致报错

joesdu, 2022-01-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
Nano pi R2S 
A：

### (3) 详细日志
Pseudo file "/dev" exists in source filesystem "/home/joe/r2s/build_dir/target-aarch64_generic_musl/root-rockchip/dev".
Ignoring, exclude it (-e/-ef) to override.
[====================================/                                                                                                                                                                                                                                                                                                                                              ]  210/2105   9%make[5]: *** [/home/joe/r2s/include/image.mk:336: /home/joe/r2s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/root.ext4] Segmentation fault (core dumped)
make[5]: *** Deleting file '/home/joe/r2s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/root.ext4'
make[5]: *** Waiting for unfinished jobs....
[==================================================================================================================================================================================================================================================================================================================================================================================/] 2105/2105 100%

Exportable Squashfs 4.0 filesystem, xz compressed, data block size 1048576
        compressed data, compressed metadata, compressed fragments,
        no xattrs, compressed ids
        duplicates are removed
Filesystem size 47899.80 Kbytes (46.78 Mbytes)
        28.73% of uncompressed filesystem size (166743.40 Kbytes)
Inode table size 17102 bytes (16.70 Kbytes)
        20.82% of uncompressed inode table size (82160 bytes)
Directory table size 23662 bytes (23.11 Kbytes)
        42.10% of uncompressed directory table size (56199 bytes)
Number of duplicate files found 308
Number of inodes 2481
Number of files 1994
Number of fragments 30
Number of symbolic links  298
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 188
Number of ids (unique uids + gids) 1
Number of uids 1
        root (0)
Number of gids 1
        root (0)
make[5]: Leaving directory '/home/joe/r2s/target/linux/rockchip/image'
make[4]: *** [Makefile:25: install] Error 2
make[4]: Leaving directory '/home/joe/r2s/target/linux/rockchip'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/joe/r2s/target/linux'
time: target/linux/install#72.95#6.52#46.88
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/home/joe/r2s'
make[1]: *** [target/Makefile:19: /home/joe/r2s/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/joe/r2s'
make: *** [/home/joe/r2s/include/toplevel.mk:230: world] Error 2
A：

coolsnowwolf, 2022-01-25:
固件占用空间过大

closed 2022-01-27 by AmadeusGhost

joesdu, 2022-01-28:
> 固件占用空间过大

但是我没有选择什么特别多的插件啊,就简单的几个常用插件.

joesdu, 2022-01-28:
> 固件占用空间过大

重新clone了代码就好了