coolsnowwolf/lede #8775: 新路由3无法搜索到WIFI信号

LemoFire, 2022-01-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
新路由3无法搜索到WIFI信号，
新编译，尝试过闭源驱动，开源驱动，和默认配置，WIFI均没有信号，希望大佬可以帮忙解答一下
CI配置：https://github.com/LemoFire/Newifi3-OpenWrt-CI/blob/f1d02f4d6b7ce0ed4333fc50df7075b62fbe107a/.github/workflows/NEWIFI-D2.yml
A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志
```
[    0.000000] Linux version 5.4.173 (lemofire@lemofire-virtual-machine) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4117-94e7b85e4)) #0 SMP Tue Jan 25 04:29:51 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Newifi-D2
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 14 pages/cpu s26992 r8192 d22160 u57344
[    0.000000] pcpu-alloc: s26992 r8192 d22160 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=000060e0
[    0.000000] Readback ErrCtl register=000060e0
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 509364K/524288K available (6170K kernel code, 256K rwdata, 1292K rodata, 1292K init, 238K bss, 14924K reserved, 0K cma-reserved, 262144K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000010] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.007824] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.073992] pid_max: default: 32768 minimum: 301
[    0.078809] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.086009] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.096819] rcu: Hierarchical SRCU implementation.
[    0.102223] smp: Bringing up secondary CPUs ...
[... 801 lines cut ...]
[  106.495769] Min Dirty = 0
[  106.498373] ExChannel = 0 , 0
[  106.501320] BW        = 80
[  106.504185] mt76x2_bbp_adjust():rf_bw=2, ext_ch=1, PrimCh=36, HT-CentCh=38, VHT-CentCh=42
[  106.558418] APStartUp(): AP Set CentralFreq at 42(Prim=36, HT-CentCh=38, VHT-CentCh=42, BBP_BW=2)
[  106.577335] mt76x2_calibration(channel = 42)
[  106.947478] Main bssid = f1:cc:20:22:c9:b2
[  106.951560] mt76x2_reinit_agc_gain:original agc_vga0 = 0x42, agc_vga1 = 0x42
[  106.958586] mt76x2_reinit_agc_gain:updated agc_vga0 = 0x42, agc_vga1 = 0x42
[  106.965528] mt76x2_reinit_hi_lna_gain:original hi_lna0 = 0x36, hi_lna1 = 0x36
[  106.972626] mt76x2_reinit_hi_lna_gain:updated hi_lna0 = 0x36, hi_lna1 = 0x36
[  106.979650] original vga value(chain0) = 42
[  106.983811] original vga value(chain1) = 42
[  106.987994] <==== rt28xx_init, Status=0
[  106.991819] RTMPDrvOpen(1):Check if PDMA is idle!
[  106.996509] RTMPDrvOpen(2):Check if PDMA is idle!
[  107.001190] WiFi Interface Up Time (rai0): 5.090s
[  107.046442] andes_pci_erasefw
[  107.069448] ==>rlt_wlan_chip_onoff(): OnOff:0, Reset= 0, pAd->WlanFunCtrl:0x20b, Reg-WlanFunCtrl=0x20b
[  107.082256] WiFi Interface Down Time (rai0): 0.070s
[  267.448936] ralink_soc_eth 1e100000.ethernet eth0: port 4 link up
[  292.905999] ralink_soc_eth 1e100000.ethernet eth0: port 4 link down
[  295.728778] ralink_soc_eth 1e100000.ethernet eth0: port 2 link up
```
A：

Boos4721, 2022-01-28:
[image]

LemoFire, 2022-01-29:
> [image]
我的搜不到 换别人编译的才行

Cathgao, 2022-01-29:
是不是没编译WiFi驱动

LemoFire, 2022-02-08:
> 是不是没编译WiFi驱动

编译了的，可以看CI配置

zhoujinshi, 2022-04-17:
lede以外其他的分支可以openwrt-immortalwrt，要对比下参数
hostapd-common - 2020-06-08-5a8b3662-40
kmod-mt76-core - 5.4.182+2021-12-03-678071ef-4
kmod-mt7603 - 5.4.182+2021-12-03-678071ef-4
kmod-mt76x2 - 5.4.182+2021-12-03-678071ef-4
luci-app-guest-wifi - 1.0
netifd - 2021-10-30-8f82742c-1
wireless-regdb - 2021.08.28-1
开启这些即可完成编译

closed 2023-02-07 by LemoFire