coolsnowwolf/lede #8781: N6005主机反复不定时自动重启

zkevinzh, 2022-01-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
系统不定时自动重启

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
N6005 cpu主机  双RTL8125 2.5g网卡
原来试过物理机装Win10 LTSC上运行hyper-v虚拟机装openwrt，现在是在PVE上运行。
情况依旧，最长2天，短的几小时就自动软重启。看过曾经有说sfe可能导致重启，编译去掉turboacc插件不能解决。
尝过以前在J1900软路由上稳定运行的旧固件同样重启，故推测可能是11代cpu N6005问题？

### (2) 路由器型号和固件版本

A：
从内核5.4.13x至现在最新5.4.173内核
固件版本OpenWrt R21.12.1、OpenWrt R22.1.1

### (3) 详细日志

A：尝试导出日志，未发现重启前有异常日志。

coolsnowwolf, 2022-01-27:
新CPU请用 5.15 内核

Wilson-coder, 2022-01-28:
我前一段时间也是遇到类似问题，后来发现“也许”是因为我更新固件时都是选择保留设置，导致/etc目录里面太多以前的文件和运行脚本，ssh登录进去把不需要的都删除掉之后，最近几天稳定下来了。目前内核用的是5.10.xx

hcym, 2022-02-12:
应该是兼容问题，cpu也是6005
我的单口pn41用大雕的和官方master都没有重启的，非常稳定，就是8125老驱动会断网，新的驱动好像没有什么事，唯一一个问题是不能安装桌面系统，win10，11和Ubuntu2004都不行，过了夏天再看看，😂

hcym, 2022-02-12:
master内核是5.4吧，用的也可以

zkevinzh, 2022-02-12:
> 应该是兼容问题，cpu也是6005 我的单口pn41用大雕的和官方master都没有重启的，非常稳定，就是8125老驱动会断网，新的驱动好像没有什么事，唯一一个问题是不能安装桌面系统，win10，11和Ubuntu2004都不行，过了夏天再看看，😂

我也正是PN41，不过还从m2转minipcie拓展多了一个8125的2.5g口，不知是不是这个的问题.....
win10LTSC我之前装过也没问题呀，就用uefi的pe一下就装上了。
还有请教一下8125新驱动是要怎么弄的？

hcym, 2022-02-12:
就是大雕库里的，最新版，移植到官方的编译可以用，用大雕源码编译更没有问题。这玩意用单口最好，我的1900是双口的还是习惯配置成单臂用

zkevinzh, 2022-02-27:
经过几周一顿测试，发现物理机运行一点问题都没有，跟cpu或者我拓展的网卡也没关系。内核用的5.4。
只要用虚拟机就有问题。exsi没试过。折腾了好久，先暂时物理机用着先了

huangzaiguo, 2022-05-31:
我也是同样的问题

GreatMichaelLee, 2022-06-28:
> 我前一段时间也是遇到类似问题，后来发现“也许”是因为我更新固件时都是选择保留设置，导致/etc目录里面太多以前的文件和运行脚本，ssh登录进去把不需要的都删除掉之后，最近几天稳定下来了。目前内核用的是5.10.xx

请问楼主这个问题解决没有，这个问题最近也很折腾我，我的还不是6005，是N5105, 我发现只要用5.15的内核，在pve上24小时之内，openwrt必然软重启，但虚拟机没事，不清任何配置，换成5.10的内核，稳定性要好很多，可以撑个两三天，但发现还是会有偶发重启，关键是根本不知道触发条件是什么，随机的。跟配置应该没关系，我觉得是openwrt内核与PVE的兼容性的问题，或者网卡直通的问题，我试过吧机型从q35改为i440fx，扩大内存等方式，统统没用 @zkevinzh @Wilson-coder @coolsnowwolf

zycboss, 2022-07-19:
N5105虚拟机，2.5G网卡，下载流量一大一会就自动重启了，日志看不到任何信息，虚拟机控制台也没有任何提示显示正常运行，只是软重启
试过5.4，5.10，5.15内核，全都有这个问题，关掉各种流量分载也无济于事，不知道是不是真是11代硬件的问题

fishg, 2022-08-02:
N5105 hyper-v 上其他虚拟机kernel panic拍到一个日志，不知道有没有参考价值
[image]

GreatMichaelLee, 2022-08-10:
> N5105虚拟机，2.5G网卡，下载流量一大一会就自动重启了，日志看不到任何信息，虚拟机控制台也没有任何提示显示正常运行，只是软重启 试过5.4，5.10，5.15内核，全都有这个问题，关掉各种流量分载也无济于事，不知道是不是真是11代硬件的问题

我跟你一模一样，看运气，有时候能坚持个几天，有时候一天几次，所有内核都试过。

zycboss, 2022-08-11:
> > N5105虚拟机，2.5G网卡，下载流量一大一会就自动重启了，日志看不到任何信息，虚拟机控制台也没有任何提示显示正常运行，只是软重启 试过5.4，5.10，5.15内核，全都有这个问题，关掉各种流量分载也无济于事，不知道是不是真是11代硬件的问题
> 
> 我跟你一模一样，但是我的应该不是流量大的问题，没有什么规律，有时候可能是半夜里，感觉触发是随机的，看运气，有时候能坚持个几天，有时候一天几次，所有内核都试过。

目前发现的规律是PVE虚拟机对2.5G网卡的直通/半虚拟化有bug，大部分人表示换成EXSi就没有这个问题了，我手上是个NAS不能装EXSi，你可以试试看。

GreatMichaelLee, 2022-08-29:
这个问题貌似无解了，我的N5105从来没有支撑超过3天的在PVE下，无论哪个版本内核（包括5.19）

zycboss, 2022-08-29:
> 这个问题貌似无解了，我的N5105从来没有支撑超过3天的在PVE下，无论哪个版本内核（包括5.19）

改旁路由吧，我现在PVE下旁路由已经稳定大半个月没出问题了

1064298219, 2022-09-19:
esxi 5105 op旁路由也是经常不定时重启

iKira, 2023-01-24:
N5105硬件的锅，换J4125保平安。

xxp2277, 2023-01-30:
> 

同样问题来报道一下。大流量下openwrt会重启（没网，然后等一会儿进openwrt发现运行时间重置了），pve管理界面中看不出openwrt重启了。pve运行日志没发现异常。

shabby2333, 2023-02-04:
5105 kernel 5.15 ove7.3 大概一天重启好几次 蹲一个解决方案

WithdewHua, 2023-02-07:
> 5105 kernel 5.15 ove7.3 大概一天重启好几次 蹲一个解决方案

之前在 v2ex 上看到的，升级内核版本可解，我是 N6005，pve 升级内核至 5.19 以上之后就正常了，目前已经稳定三个月了

shabby2333, 2023-02-08:
> > 5105 kernel 5.15 ove7.3 大概一天重启好几次 蹲一个解决方案
> 
> 之前在 v2ex 上看到的，升级内核版本可解，我是 N6005，pve 升级内核至 5.19 以上之后就正常了，目前已经稳定三个月了

十分感谢，前两天我也看到了这篇，目前没发现重启现象~

coolsnowwolf, 2026-09-05:
N6005/PVE 随机重启报告没有崩溃日志、硬件错误或当前版本复现证据，且测试版本为 2021/2022 年旧内核；关闭为 not planned。

closed 2026-09-05 by coolsnowwolf