coolsnowwolf/lede #8787: 路由器重启后wifi不会默认打开

Jinnrry, 2022-01-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：路由器只要重启以后，无线网络不会默认打开，必须手动开启一次无线网络

### (2) 路由器型号和固件版本

A：新3 OpenWrt R22.1.1 / LuCI Master (git-21.335.48743-5f363d9)

### (3) 详细日志

A：
```
[    0.000000] Linux version 5.4.162 (jiangwe@DESKTOP-8K03O5L) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4079-60e275903)) #0 SMP Thu Jan 13 02:11:31 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Newifi-D2
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 14 pages/cpu s26992 r8192 d22160 u57344
[    0.000000] pcpu-alloc: s26992 r8192 d22160 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00008c10
[    0.000000] Readback ErrCtl register=00008c10
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 509428K/524288K available (6135K kernel code, 254K rwdata, 1288K rodata, 1268K init, 238K bss, 14860K reserved, 0K cma-reserved, 262144K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.007825] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.073989] pid_max: default: 32768 minimum: 301
[    0.078805] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.086006] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.096800] rcu: Hierarchical SRCU implementation.
[    0.102208] smp: Bringing up secondary CPUs ...
[    0.364270] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[... 1228 lines cut ...]
[  286.202102] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  287.242028] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  288.282019] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  290.712144] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  291.722134] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  292.761908] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  293.801816] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  294.841763] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  297.311939] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  298.361757] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  299.401759] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  300.441762] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  301.481774] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  303.911970] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  304.921830] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  305.961835] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  307.001863] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  308.041839] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  310.511938] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  311.561750] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  312.601763] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  313.641754] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00
[  314.681782] AP-Client probe: SSID=, BSSID=00:00:00:00:00:00

```

nanchuci, 2022-01-29:
闭源驱动，你是不是改了LAN口设置

Jinnrry, 2022-01-29:
> 闭源驱动，你是不是改了LAN口设置

没动呀

kzkzzzz, 2022-01-30:
暂时手动开启可以解决，在启动项脚本加上
```bash
# 启动2.4g 和 5g 信号
ip link set ra0 up
ip link set rai0 up

# 桥接网卡
brctl addif br-lan ra0
brctl addif br-lan rai0
```

liwenjie119, 2022-02-04:
lan口改为192.168.100.3然后新三lan口有线接上级路由器当ap使用，重启后wifi无法启动，需要在页面重启无线。

hnrrn, 2022-02-10:
> 闭源驱动，你是不是改了LAN口设置

我是在编译的时候更改了lan 口ip ，也出现这个问题。。。难道改
LAN口设置会影响的无线启动吗

Jinnrry, 2022-02-10:
@hnrrn  @liwenjie119  我也是改了Lan口IP，就会出这个问题，按照kzkzzzz说的，自己加了一个启动项解决了

脚本内容

```
#!/bin/sh /etc/rc.common
START=99
start() {
  ip link set ra0 up
  ip link set rai0 up

  brctl addif br-lan ra0
  brctl addif br-lan rai0
}

```

放到/etc/init.d/start_wifi 然后执行 `/etc/init.d/start_wifi enable`

modelsun, 2023-03-21:
> Jinnrr

> 

请问这个怎么加到源码里，编译进固件？能这样搞吗？谢谢

coolsnowwolf, 2026-09-05:
评论已确认问题与 MT7621 闭源无线驱动及修改 LAN/AP 配置后的接口初始化有关，并提供了手工拉起 ra0/rai0 的 init workaround；当前驱动与目标栈已变化，不能安全地把该 workaround 作为通用源码修复。

closed 2026-09-05 by coolsnowwolf