coolsnowwolf/lede #8789: ps3netsrv[4771]: segfault at 400040 ip 0000000000418fbf sp 00007fff24e3f800 error 4 in ps3netsrv[401000+1e000]

atois, 2022-01-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
在控制台配置ps3netsrv后，服务无法启动

### (1) 具体问题

排查过程

log
Sat Jan 29 13:39:24 2022 daemon.info ps3netsrv: ps3netsrv will run as user 'root'.
Sat Jan 29 13:39:24 2022 daemon.info procd: Instance ps3netsrv::ps3netsrv.main s in a crash loop 9 crashes, 0 seconds since last crash
Sat Jan 29 13:39:24 2022 kern.info kernel: [391404.090787] ps3netsrv[18747]: segfault at 400040 ip 0000000000418fbf sp 00007ffc142b4e10 error 4 in ps3netsrv[401000+1e000]
Sat Jan 29 13:39:24 2022 kern.info kernel: [391404.091191] Code: 44 8b 05 b0 e2 00 00 45 31 d2 31 c9 55 31 d2 bd 00 00 80 00 53 4c 8b 5f 18 bb 00 00 00 00 4c 8b 4f 28 4c 89 d8 4d 85 c9 74 59 <8b> 30 83 fe 06 75 05 4c 89 d9 eb 14 83 fe 02 74 07 83 fe 07 74 37

root@OpenWrt:~# dmesg | grep ps3netsrv
[341409.719276] ps3netsrv[28838]: segfault at 400040 ip 0000000000418fbf sp 00007fff933ebd30 error 4 in ps3netsrv[401000+1e000]
[341414.721429] ps3netsrv[28911]: segfault at 400040 ip 0000000000418fbf sp 00007ffd1e3cc300 error 4 in ps3netsrv[401000+1e000]
[341419.307187] ps3netsrv[29033]: segfault at 400040 ip 0000000000418fbf sp 00007ffdb066f3a0 error 4 in ps3netsrv[401000+1e000]
[341424.309069] ps3netsrv[29127]: segfault at 400040 ip 0000000000418fbf sp 00007ffd79169e90 error 4 in ps3netsrv[401000+1e000]
[341429.311736] ps3netsrv[29195]: segfault at 400040 ip 0000000000418fbf sp 00007ffd36f51a10 error 4 in ps3netsrv[401000+1e000]
[341429.845518] ps3netsrv[29280]: segfault at 400040 ip 0000000000418fbf sp 00007ffe55693b80 error 4 in ps3netsrv[401000+1e000]
[341443.647304] ps3netsrv[29493]: segfault at 400040 ip 0000000000418fbf sp 00007ffd451c45b0 error 4 in ps3netsrv[401000+1e000]
[341458.718515] ps3netsrv[29725]: segfault at 400040 ip 0000000000418fbf sp 00007ffd1264dbd0 error 4 in ps3netsrv[401000+1e000]
[387188.969076] ps3netsrv[18860]: segfault at 400040 ip 0000000000418fbf sp 00007ffece802940 error 4 in ps3netsrv[401000+1e000]
[387262.030143] ps3netsrv[19597]: segfault at 400040 ip 0000000000418fbf sp 00007ffed0f38ef0 error 4 in ps3netsrv[401000+1e000]
[387314.596519] ps3netsrv[20120]: segfault at 400040 ip 0000000000418fbf sp 00007fff62bb4650 error 4 in ps3netsrv[401000+1e000]
[389301.035004] ps3netsrv[32051]: segfault at 400040 ip 0000000000418fbf sp 00007ffe636ca810 error 4 in ps3netsrv[401000+1e000]
[389918.224830] ps3netsrv[4475]: segfault at 400040 ip 0000000000418fbf sp 00007ffeee0c1160 error 4 in ps3netsrv[401000+1e000]
[389923.226644] ps3netsrv[4567]: segfault at 400040 ip 0000000000418fbf sp 00007fffdae8e850 error 4 in ps3netsrv[401000+1e000]
[389928.229490] ps3netsrv[4631]: segfault at 400040 ip 0000000000418fbf sp 00007fffddef6510 error 4 in ps3netsrv[401000+1e000]
[389933.232364] ps3netsrv[4678]: segfault at 400040 ip 0000000000418fbf sp 00007ffc891034f0 error 4 in ps3netsrv[401000+1e000]
[389938.235141] ps3netsrv[4724]: segfault at 400040 ip 0000000000418fbf sp 00007fff66f50300 error 4 in ps3netsrv[401000+1e000]
[389943.237999] ps3netsrv[4771]: segfault at 400040 ip 0000000000418fbf sp 00007fff24e3f800 error 4 in ps3netsrv[401000+1e000]
[390537.908983] ps3netsrv[10619]: segfault at 400040 ip 0000000000418fbf sp 00007ffd7b266870 error 4 in ps3netsrv[401000+1e000]
[390679.024144] ps3netsrv[11943]: segfault at 400040 ip 0000000000418fbf sp 00007fffc7cad190 error 4 in ps3netsrv[401000+1e000]
[391404.090787] ps3netsrv[18747]: segfault at 400040 ip 0000000000418fbf sp 00007ffc142b4e10 error 4 in ps3netsrv[401000+1e000]

官方解释
In computing, a segmentation fault (often shortened to segfault) or access violation is a fault, or failure condition, raised by hardware with memory protection, notifying an operating system (OS) the software has attempted to access a restricted area of memory (a memory access violation).

segfault at和error 4这两条信息可以得出是内存读出错，4的意义如下，可以对照参考：
bit2:值为1表示是用户态程序内存访问越界，值为0表示是内核态程序内存访问越界
bit1: 值为1表示是写操作导致内存访问越界，值为0表示是读操作导致内存访问越界
bit0: 值为1表示没有足够的权限访问非法地址的内容，值为0表示访问的非法地址根本没有对应的页面，也就是无效地址

4（100）为用户态程序内存访问越界。

正在查看以下代码
https://gitce.net/mirrors/lede/commit/340f1bffaa5ef9700490beb257b6132ae516c4e2

继续排查中，待续·····

A：
### (2) 路由器型号和固件版本
路由器型号
AMD GX-420CA SOC with Radeon(tm) HD Graphics : 4 Core 4 Thread

固件版本
OpenWrt R21.6.1 GDQ v6.1[2021] Compiled by eSir / LuCI Master (git-21.114.56541-974fb04)

WYC-2020, 2022-02-07:
我这边x86 一切正常
[image]

coolsnowwolf, 2026-09-06:
ps3netsrv 反复 segfault 缺少版本、配置和可复现调用栈，属于独立第三方程序问题，关闭为“不计划处理”。

closed 2026-09-06 by coolsnowwolf