coolsnowwolf/lede #8790: qBitorrent无法运行

pacinoa, 2022-01-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
qBitorrent无法运行
A：

### (2) 路由器型号和固件版本

A： 
<html>
<body>

固件版本 | OpenWrt R22.1.1 / LuCI Master (git-21.335.48743-5f363d9)
-- | --
内核版本 | 5.15.16

</body>
</html>
### (3) 详细日志

A：at Jan 29 16:34:22 2022 kern.info kernel: [265578.653082] traps: qbittorrent-nox[9296] general protection fault ip:7fa79fc459eb sp:7fa79eb6b878 error:0 in libc.so[7fa79fc36000+49000]
Sat Jan 29 16:34:29 2022 kern.info kernel: [265585.491462] traps: qbittorrent-nox[23802] general protection fault ip:7f3b8b6289ea sp:7f3b8a546b78 error:0 in libc.so[7f3b8b619000+49000]
Sat Jan 29 16:34:35 2022 kern.info kernel: [265591.750183] traps: qbittorrent-nox[23937] general protection fault ip:7f689c1279eb sp:7f689b045b78 error:0 in libc.so[7f689c118000+49000]
Sat Jan 29 16:34:41 2022 kern.info kernel: [265597.974504] traps: qbittorrent-nox[24064] general protection fault ip:7f5c64c989eb sp:7f5c63bb6b78 error:0 in libc.so[7f5c64c89000+49000]
Sat Jan 29 16:34:47 2022 kern.info kernel: [265604.209455] traps: qbittorrent-nox[24151] general protection fault ip:7feb94b209ea sp:7feb93a3eb78 error:0 in libc.so[7feb94b11000+49000]
Sat Jan 29 16:34:54 2022 kern.info kernel: [265610.436145] traps: qbittorrent-nox[24255] general protection fault ip:7fad2eb8a9eb sp:7fad2daa8b78 error:0 in libc.so[7fad2eb7b000+49000]
Sat Jan 29 16:35:01 2022 kern.info kernel: [265617.668401] traps: qbittorrent-nox[24365] general protection fault ip:7f6ca8e069eb sp:7f6ca7d24878 error:0 in libc.so[7f6ca8df7000+49000]
Sat Jan 29 16:35:01 2022 daemon.info procd: Instance qbittorrent::qbittorrent.main s in a crash loop 6 crashes, 3 seconds since last crash

WYC-2020, 2022-01-29:
是一下载东西就崩溃？这个我以前分析过，暂时回滚boost到1.76吧，我这边测试没问题，你也可以找下类似的issue，有一个

pacinoa, 2022-01-29:
> 是一下载东西就崩溃？这个我以前分析过，暂时回滚boost到1.76吧，我这边测试没问题，你也可以找下类似的issue，有一
请问回滚版本如何操作?

lxl6125, 2022-01-30:
> 

用预编译文件 qbittorrent_static 吧，确定能正常运行

540240126, 2022-02-07:
@WYC-2020 大佬请问下回滚怎么操作，我也遇到这个问题了，感谢

WYC-2020, 2022-02-07:
@540240126 明天给你链接

WYC-2020, 2022-02-08:
@540240126 https://github.com/WYC-2020/packages/commit/c63a3935a79f597c49af1b3f2ba193406a201c22

540240126, 2022-02-08:
@WYC-2020 好的谢谢