coolsnowwolf/lede #8791: K2P 每隔不到一天，就会自动重启，大概17-20小时

ddd-zero, 2022-01-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：K2P A1 
固件版本	OpenWrt R22.1.1 / LuCI Master (git-21.335.48743-5f363d9)
内核版本	5.4.174
运行时间	0h 13m 13s
### (3) 详细日志

A：
Sat Jan 29 23:16:35 2022 kern.warn kernel: [   28.781547] BSSAifsn[1]=7
Sat Jan 29 23:16:35 2022 kern.warn kernel: [   28.786848] BSSAifsn[2]=2
Sat Jan 29 23:16:35 2022 kern.warn kernel: [   28.792066] BSSAifsn[3]=2
Sat Jan 29 23:16:35 2022 kern.warn kernel: [   28.797763] BSSAifsn[0]=3
Sat Jan 29 23:16:35 2022 kern.warn kernel: [   28.803034] BSSAifsn[1]=7
Sat Jan 29 23:16:35 2022 kern.warn kernel: [   28.808319] BSSAifsn[2]=2
Sat Jan 29 23:16:35 2022 kern.warn kernel: [   28.813605] BSSAifsn[3]=2
Sat Jan 29 23:16:35 2022 kern.warn kernel: [   28.819268] BSSAifsn[0]=3
Sat Jan 29 23:16:35 2022 kern.warn kernel: [   28.824606] BSSAifsn[1]=7
Sat Jan 29 23:16:35 2022 kern.warn kernel: [   28.829907] BSSAifsn[2]=2
Sat Jan 29 23:16:35 2022 kern.warn kernel: [   28.835253] BSSAifsn[3]=2
Sat Jan 29 23:16:35 2022 kern.warn kernel: [   28.840801] BSSAifsn[0]=3
Sat Jan 29 23:16:35 2022 kern.warn kernel: [   28.846056] BSSAifsn[1]=7
Sat Jan 29 23:16:35 2022 kern.warn kernel: [   28.851277] BSSAifsn[2]=2
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.856560] BSSAifsn[3]=2
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.862134] BSSAifsn[0]=3
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.867512] BSSAifsn[1]=7
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.872851] BSSAifsn[2]=2
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.878206] BSSAifsn[3]=2
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.883890] BSSAifsn[0]=3
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.889183] BSSAifsn[1]=7
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.894487] BSSAifsn[2]=2
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.899823] BSSAifsn[3]=2
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.907854] DfsZeroWait Support=0/0
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.916311] DfsZeroWaitCacTime=255/255
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.924780] read_etxbf: ETxBfEnCond = 0
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.932491] read_etxbf: BSSID[0]
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.939012] read_etxbf: MBSS[0] ETxBfEnCond = 0
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.948123] read_etxbf: ETxBfEnCond = 0
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.955821] read_etxbf: BSSID[1]
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.962373] read_etxbf: MBSS[1] ETxBfEnCond = 0
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.971550] read_etxbf: ETxBfEnCond = 0
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.979332] read_etxbf: BSSID[2]
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.985909] read_etxbf: MBSS[2] ETxBfEnCond = 0
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   28.995093] read_etxbf: ETxBfEnCond = 0
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   29.002812] read_etxbf: BSSID[3]
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   29.009375] read_etxbf: MBSS[3] ETxBfEnCond = 0
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   29.018548] read_etxbf: ETxBfEnCond = 1
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   29.026306] read_etxbf: BSSID[4]
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   29.032840] read_etxbf: MBSS[4] ETxBfEnCond = 1
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   29.042019] read_etxbf: ETxBfEnCond = 1
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   29.049790] read_etxbf: BSSID[5]
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   29.056302] read_etxbf: MBSS[5] ETxBfEnCond = 1
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   29.065440] read_etxbf: ETxBfEnCond = 1
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   29.073149] read_etxbf: BSSID[6]
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   29.079628] read_etxbf: MBSS[6] ETxBfEnCond = 1
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   29.088693] read_etxbf: ETxBfEnCond = 1
Sat Jan 29 23:16:36 2022 kern.warn kernel: [   29.096383] read_etxbf: BSSID[7]
[... 291 lines cut ...]
Sat Jan 29 23:54:05 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:05 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:10 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:10 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:10 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:15 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:15 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:15 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:19 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:19 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:19 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:25 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:25 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:25 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:30 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:30 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:30 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:34 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:34 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:34 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:37 2022 kern.err /usr/bin/ssr-redir[5005]: remote recv: Connection reset by peer
Sat Jan 29 23:54:39 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:39 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:39 2022 daemon.err uhttpd[1680]: assoclist ioctl fails
Sat Jan 29 23:54:45 2022 daemon.err uhttpd[1680]: assoclist ioctl fails

ddd-zero, 2022-01-29:
大概是23:40断的

ddd-zero, 2022-02-02:
应该就是开了HWNAT导致的重启。我关闭了之后就没有问题了。已经连续运行3天了。等过一个星期还是稳定的话，我再测试一下，关闭BBR,开启HWNAT。看看是否是这俩冲突导致的

xixky, 2022-02-17:
我小米mini也出现这个问题，打开硬件加速几小时就重启一次

xixky, 2022-02-17:
我现在关闭硬件加速还在观察会不会重启

xixky, 2022-02-17:
可能mt7620性能太垃圾了，关闭硬件加速，cpu占用还是高

xixky, 2022-02-17:
[image]

ddd-zero, 2022-02-17:
> [image]

我7621 关闭硬件，开ssr，NAS上跑PT . CPU占用个位数百分百，负载0.1左右

xiaopang0v0, 2022-10-27:
> 我小米mini也出现这个问题，打开硬件加速几小时就重启一次

其实不是硬件加速的问题，是 WIFI 的问题，你只要不连你的 WIFI，MT7620 可以很长时间不自动重启的