coolsnowwolf/lede #8793: 官方内核更新了。5.15.18

baodp, 2022-01-30:
您们好。要不要更新下内核5.15.18？？

WYC-2020, 2022-01-30:
那个官方给个链接我看看哈哈

AmadeusGhost, 2022-01-30:
不用过年吗？

Jason6111, 2022-01-30:
> 那个官方给个链接我看看哈哈

http://mirrors.ustc.edu.cn/kernel.org/linux/kernel/v5.x/  
https://www.kernel.org/  
让大佬们好好休息过节吧

Myoko, 2022-01-30:
大过年的！消停一下吧

AXBNG, 2022-01-30:
大过年的没事做还在这儿追新刷机？？？

baodp, 2022-01-30:
大家新年好，谢谢你们的热烈回复。祝大家恭喜发财，身体健康！！

baodp, 2022-01-30:
> 大过年的没事做还在这儿追新刷机？？？

中年老男人无聊啊。只有刷机这乐趣了。。。。

baodp, 2022-01-30:
> 中年老男人无聊啊。只有刷机这乐趣了。。。。

中年老男人无聊啊。只有刷机这乐趣了。。。。

lxl6125, 2022-01-30:
自己改一下不就行了？
[image]
[image]

baodp, 2022-01-30:
> 自己改一下不就行了？ [image] [image]

怎么弄的？？

Jason6111, 2022-01-31:
替换下就可以 但补丁…

Jason6111, 2022-01-31:
> > 自己改一下不就行了？ [image] [image]
> 
> 怎么弄的？？

include/kernel-5.15 改这里 没补丁不好用就别说什么

moonerick, 2022-02-01:
> > > 自己改一下不就行了？ [image] [image]
> > 
> > 
> > 怎么弄的？？
> 
> include/kernel-5.15 改这里 没补丁不好用就别说什么

请问这个hash值是从https://www.kernel.org/的那个位置获得的呢

closed 2022-02-01 by AmadeusGhost

Karmylr, 2022-02-01:
可以编译5.15.18了？