coolsnowwolf/lede #8798: Frpc关闭【开启网页管理】报错

RealKiro, 2022-01-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
Frpc关闭`开启网页管理`报错，提示：
```markdown
一个或多个必选项值为空！
一个或多个必选项值为空！
一个或多个必选项值为空！
```
[image]

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue，问题可复现

## 2. 详细叙述

### (1) 具体问题

Q：复现步骤：
1. 开启网页管理 ☑️-- 保存应用
2. 开启网页管理 取消☑️ -- 保存应用 --报错
```markdown
一个或多个必选项值为空！
一个或多个必选项值为空！
一个或多个必选项值为空！
```

### (2) 路由器型号和固件版本

A：R619ac 128M /  (git-21.335.48743-5f363d9) / OpenWrt R22.1.1

### (3) 详细日志

`/var/logs/frps.log` 无报错信息，以下是系统日志：
```markdown
Mon Jan 31 16:59:44 2022 user.notice FRPC: Shutting down frp service
Mon Jan 31 16:59:45 2022 user.notice FRPC: Starting frp service
Mon Jan 31 16:59:47 2022 kern.err kernel: [ 7007.889646] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 16:59:52 2022 kern.err kernel: [ 7012.209577] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 16:59:56 2022 kern.err kernel: [ 7016.549594] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 17:00:00 2022 kern.err kernel: [ 7020.869168] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 17:00:00 2022 cron.err crond[11725]: USER root pid 26577 cmd /etc/init.d/frp restart
Mon Jan 31 17:00:00 2022 user.notice FRPC: Shutting down frp service
Mon Jan 31 17:00:01 2022 user.notice FRPC: Starting frp service
Mon Jan 31 17:00:04 2022 kern.err kernel: [ 7025.189571] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 17:00:09 2022 kern.err kernel: [ 7029.509542] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 17:00:13 2022 kern.err kernel: [ 7033.839522] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 17:00:17 2022 kern.err kernel: [ 7038.179512] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 17:00:22 2022 kern.err kernel: [ 7042.509551] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 17:00:26 2022 kern.err kernel: [ 7046.829556] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 17:00:30 2022 kern.err kernel: [ 7051.149527] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 17:00:35 2022 kern.err kernel: [ 7055.469562] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 17:00:39 2022 kern.err kernel: [ 7059.809508] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 17:00:43 2022 kern.err kernel: [ 7064.129525] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 17:00:44 2022 user.notice FRPC: Shutting down frp service
Mon Jan 31 17:00:45 2022 user.notice FRPC: Starting frp service
Mon Jan 31 17:00:48 2022 kern.err kernel: [ 7068.449512] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 17:00:52 2022 kern.err kernel: [ 7072.769550] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 17:00:56 2022 kern.err kernel: [ 7077.109470] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 17:01:01 2022 kern.err kernel: [ 7081.439031] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 17:01:05 2022 kern.err kernel: [ 7085.759421] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 17:01:09 2022 kern.err kernel: [ 7090.079444] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 17:01:14 2022 kern.err kernel: [ 7094.409525] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
Mon Jan 31 17:01:18 2022 user.notice FRPC: Shutting down frp service
Mon Jan 31 17:01:18 2022 user.notice FRPC: Starting frp service
```

WYC-2020, 2022-02-05:
https://github.com/WYC-2020/lede/commit/424807ecddf7ae868b0d195117190ea1e021d66e
@RealKiro

Beginner-Go, 2022-02-07:
麻烦大佬提交下pr

RealKiro, 2022-02-12:
@WYC-2020  @Beginner-Go  `FRPS`关闭日志的选项的时候也存在这个问题
[image]

Q：复现步骤：

启用日志 ☑️-- 保存应用
启用日志 取消☑️ -- 保存应用 --报错
```markdown
一个或多个必选项值为空！
```

Beginner-Go, 2022-02-12:
选项全试试吧 看看还有什么bug

WYC-2020, 2022-02-12:
fprs 是另外一个 我没看 @RealKiro

WYC-2020, 2022-02-12:
https://github.com/WYC-2020/lede/commit/589245ae81b828241726680397282d3c75c99632
@RealKiro  试下这个，我发现最近很多出这样的问题哦，是更新了啥基础库导致还是这些库被人更新了，以前都没人反馈

Beginner-Go, 2022-02-12:
可能这些插件用的人少

RealKiro, 2022-02-12:
> 选项全试试吧 看看还有什么bug

选项都试了，除了这个bug外应该是没有了

closed 2022-02-12 by Beginner-Go