coolsnowwolf/lede #8820: R4S 可以编译5.15内核吗

GreatMichaelLee, 2022-02-02:
如题，好像make menuconfig设置using testing kernel 也米用啊。强制改.config里把CONFIG_LINUX_5_10=y 改成CONFIG_LINUX_5_15=y 然后改./target/linux/rockchip/Makefile 里面为KERNEL_TESTING_PATCHVER = 5.15进行编译，最后还是报错啊。是R4S还没有开始做兼容适配吗？

Ra2-IFV, 2022-02-02:
不行，因为还没上

jxph1123, 2022-02-02:
5.10都还没有正常。。

GreatMichaelLee, 2022-02-03:
> 5.10都还没有正常。。

5.10我在R4S上已经稳定跑了个把月了，应该是你自己的问题吧

jxph1123, 2022-02-03:
> > 5.10都还没有正常。。
> 
> 5.10我在R4S上已经稳定跑了个把月了，应该是你自己的问题吧

5.10是不是不能恢复交换机，我恢复了交换机不正常。。

GreatMichaelLee, 2022-02-03:
> > > 5.10都还没有正常。。
> > 
> > 
> > 5.10我在R4S上已经稳定跑了个把月了，应该是你自己的问题吧
> 
> 5.10是不是不能恢复交换机，我恢复了交换机不正常。。

啥叫恢复交换机？你是装在虚拟机上吗

jxph1123, 2022-02-03:
> > > > 5.10都还没有正常。。
> > > 
> > > 
> > > 5.10我在R4S上已经稳定跑了个把月了，应该是你自己的问题吧
> > 
> > 
> > 5.10是不是不能恢复交换机，我恢复了交换机不正常。。
> 
> 啥叫恢复交换机？你是装在虚拟机上吗
swconfig

1989car, 2022-02-07:
> > 5.10都还没有正常。。
> 
> 5.10我在R4S上已经稳定跑了个把月了，应该是你自己的问题吧

using testing kernel 编译出来的是5.10吧，不知道为啥用5.10的固件R4S启动不起来，sys灯一直闪，两个网口的灯不亮。

GreatMichaelLee, 2022-02-08:
> > > 5.10都还没有正常。。
> > 
> > 
> > 5.10我在R4S上已经稳定跑了个把月了，应该是你自己的问题吧
> 
> using testing kernel 编译出来的是5.10吧，不知道为啥用5.10的固件R4S启动不起来，sys灯一直闪，两个网口的灯不亮。

make clean再编译

1989car, 2022-02-08:
> > > > 5.10都还没有正常。。
> > > 
> > > 
> > > 5.10我在R4S上已经稳定跑了个把月了，应该是你自己的问题吧
> > 
> > 
> > using testing kernel 编译出来的是5.10吧，不知道为啥用5.10的固件R4S启动不起来，sys灯一直闪，两个网口的灯不亮。
> 
> make clean再编译

谢谢 我再试试

closed 2022-02-12 by Beginner-Go