coolsnowwolf/lede #8821: 日志内容 ’daemon.err uhttpd[64097]: ethtool: bad command line argument(s)‘

vxhorse, 2022-02-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：查看日志时，被’daemon.err uhttpd[64097]: ethtool: bad command line argument(s)‘刷屏，然后看了下进程写的是
`<html>
<body>

/usr/sbin/uhttpd -f -h /www -r OpenWrt -x /cgi-bin -u /ubus -t 60 -T 30 -k 20 -A 1 -n 3 -N 100 -R -p 0.0.0.0:80 -p [::]:80 -I cgi-bin/luci -C /etc/uhttpd.crt -K /etc/uhttpd.key -s 0.0.0.0:443 -s [::]:443
--

</body>
</html>`

### (2) 路由器型号和固件版本

A：在Ubuntu服务器中使用Docker运行Openwrt

### (3) 详细日志

A：
Wed Feb  2 21:17:04 2022 daemon.err uhttpd[64097]: For more information run ethtool -h
Wed Feb  2 21:17:04 2022 daemon.err uhttpd[64097]: sh: yes: unknown operand
Wed Feb  2 21:17:05 2022 daemon.err uhttpd[64097]: ethtool: bad command line argument(s)
Wed Feb  2 21:17:05 2022 daemon.err uhttpd[64097]: For more information run ethtool -h
Wed Feb  2 21:17:05 2022 daemon.err uhttpd[64097]: sh: yes: unknown operand
Wed Feb  2 21:17:06 2022 daemon.err uhttpd[64097]: ethtool: bad command line argument(s)
Wed Feb  2 21:17:06 2022 daemon.err uhttpd[64097]: For more information run ethtool -h
Wed Feb  2 21:17:06 2022 daemon.err uhttpd[64097]: sh: yes: unknown operand
Wed Feb  2 21:17:06 2022 daemon.err uhttpd[64097]: ethtool: bad command line argument(s)
Wed Feb  2 21:17:06 2022 daemon.err uhttpd[64097]: For more information run ethtool -h
Wed Feb  2 21:17:06 2022 daemon.err uhttpd[64097]: sh: yes: unknown operand
Wed Feb  2 21:17:07 2022 daemon.err uhttpd[64097]: ethtool: bad command line argument(s)
Wed Feb  2 21:17:07 2022 daemon.err uhttpd[64097]: For more information run ethtool -h
Wed Feb  2 21:17:07 2022 daemon.err uhttpd[64097]: sh: yes: unknown operand
Wed Feb  2 21:17:08 2022 daemon.err uhttpd[64097]: ethtool: bad command line argument(s)
Wed Feb  2 21:17:08 2022 daemon.err uhttpd[64097]: For more information run ethtool -h
Wed Feb  2 21:17:08 2022 daemon.err uhttpd[64097]: sh: yes: unknown operand

roacn, 2022-02-04:
检查下是否有以下2种情况出现：
1、openwrt的WEB管理页面，网卡接口无法识别，显示为`半双工`、网速为`-`；
2、在命令行运行`ip address`命令看下网卡信息，是否含有`@`的显示，比如`eth0@if11`；
如果有以上情况，很可能是打开WEB管理页面的总览页面时，加载`/usr/lib/lua/luci/view/admin_status/index.htm`时，代码运行`ethinfo`错误导致的。

修改：
备份`/sbin/ethinfo`，把`/sbin/ethinfo`(源码`package/lean/autocore/files/x86/sbin/ethinfo`)里获取网卡信息的代码修改一下，
`a=$(ip address | grep ^[0-9] | awk -F: '{print $2}' | sed "s/ //g" | grep '^[e]' | grep -v "@" | grep -v "\.")`
修改为
`a=$(ip address | grep ^[0-9] | awk -F: '{print $2}' | sed "s/ //g" | grep '^[e]' | awk -F"@" '{print $1}')`
在lxc容器的openwrt以上修改可以修复网卡接口信息显示以及`ethtool: bad command line argument(s)`刷屏，同时也不影响在`ip address`里显示为`eth0`的openwrt。
Docker运行Openwrt这个环境没用过，可以尝试下。

vxhorse, 2022-02-06:
> 检查下是否有以下2种情况出现： 1、openwrt的WEB管理页面，网卡接口无法识别，显示为`半双工`、网速为`-`； 2、在命令行运行`ip address`命令看下网卡信息，是否含有`@`的显示，比如`eth0@if11`； [...]

vxhorse, 2022-02-06:
> 检查下是否有以下2种情况出现： 1、openwrt的WEB管理页面，网卡接口无法识别，显示为`半双工`、网速为`-`； 2、在命令行运行`ip address`命令看下网卡信息，是否含有`@`的显示，比如`eth0@if11`； [...]

感谢大佬！你说的两种情况都出现了，按照你的方法解决了问题。

closed 2022-02-06 by vxhorse

smdx, 2022-03-02:
> 检查下是否有以下2种情况出现： 1、openwrt的WEB管理页面，网卡接口无法识别，显示为`半双工`、网速为`-`； 2、在命令行运行`ip address`命令看下网卡信息，是否含有`@`的显示，比如`eth0@if11`； [...]

大佬你好，在LXC运行的op按你的方法eth0、eth1网卡正常显示了，就是在首页多了个erspan0-半双工的虚拟网卡显示了请问怎么去除？谢谢

roacn, 2022-03-04:
> > 检查下是否有以下2种情况出现： 1、openwrt的WEB管理页面，网卡接口无法识别，显示为`半双工`、网速为`-`； 2、在命令行运行`ip address`命令看下网卡信息，是否含有`@`的显示，比如`eth0@if11`； 如果有以上情况，很可能是打开WEB管理页面的总览页面时，加载`/usr/lib/lua/luci/view/admin_status/index.htm`时，代码运行`ethinfo`错误导致的。
> > 修改： 备份`/sbin/ethinfo`，把`/sbin/ethinfo`(源码`package/lean/autocore/files/x86/sbin/ethinfo`)里获取网卡信息的代码修改一下， `a=$(ip address | grep ^[0-9] | awk -F: '{print $2}' | sed "s/ //g" | grep '^[e]' | grep -v "@" | grep -v "\.")` 修改为 `a=$(ip address | grep ^[0-9] | awk -F: '{print $2}' | sed "s/ //g" | grep '^[e]' | awk -F"@" '{print $1}')` 在lxc容器的openwrt以上修改可以修复网卡接口信息显示以及`ethtool: bad command line argument(s)`刷屏，同时也不影响在`ip address`里显示为`eth0`的openwrt。 Docker运行Openwrt这个环境没用过，可以尝试下。
> 
> 大佬你好，在LXC运行的op按你的方法eth0、eth1网卡正常显示了，就是在首页多了个erspan0 半双工的虚拟网卡显示了请问怎么去除？谢谢 另外修改后有个错误日志一直重复出现： Wed Mar 2 18:52:02 2022 daemon.err uhttpd[1220]: sh: yes: unknown operand Wed Mar 2 18:52:03 2022 daemon.err uhttpd[1220]: sh: yes: unknown operand Wed Mar 2 18:52:04 2022 daemon.err uhttpd[1220]: sh: yes: unknown operand Wed Mar 2 18:52:05 2022 daemon.err uhttpd[1220]: sh: yes: unknown operand Wed Mar 2 18:52:07 2022 daemon.err uhttpd[1220]: sh: yes: unknown operand Wed Mar 2 18:52:08 2022 daemon.err uhttpd[1220]: sh: yes: unknown operand Wed Mar 2 18:52:09 2022 daemon.err uhttpd[1220]: sh: yes: unknown operand Wed Mar 2 18:52:10 2022 daemon.err uhttpd[1220]: sh: yes: unknown operand Wed Mar 2 18:52:11 2022 daemon.err uhttpd[1220]: sh: yes: unknown operand Wed Mar 2 18:52:12 2022 daemon.err uhttpd[1220]: sh: yes: unknown operand Wed Mar 2 18:52:13 2022 daemon.err uhttpd[1220]: sh: yes: unknown operand Wed Mar 2 18:52:14 2022 daemon.err uhttpd[1220]: sh: yes: unknown operand Wed Mar 2 18:52:15 2022 daemon.err uhttpd[1220]: sh: yes: unknown operand Wed Mar 2 18:52:16 2022 daemon.err uhttpd[1220]: sh: yes: unknown operand Wed Mar 2 18:52:17 2022 daemon.err uhttpd[1220]: sh: yes: unknown operand Wed Mar 2 18:52:18 2022 daemon.err uhttpd[1220]: sh: yes: unknown operand Wed Mar 2 18:52:19 2022 daemon.err uhttpd[1220]: sh: yes: unknown operand Wed Mar 2 18:52:20 2022 daemon.err uhttpd[1220]: sh: yes: unknown operand

匹配eth的网卡
`a=$(ip address | grep ^[0-9] | awk -F: '{print $2}' | sed "s/ //g" | grep '^eth' | awk -F"@" '{print $1}')`