coolsnowwolf/lede #8823: docker自建网络父接口问题，docker固定ip问题，自建网络流量不走ssr+

Xunflash, 2022-02-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
#7384 

## 2. 详细叙述

### (1) 具体问题

A：问题一：使用dockerman创建两个容器，使用默认桥接（bridge）模式时，宿主机（OpenWrt）重启后会使两个容器分配的ip互换（默认网段是172.17.0.0/16  现假设容器A分配了172.17.0.2，容器B分配172.17.0.3，在宿主机重启后，会根据容器启动的顺序给容器分配ip，一些时候会让容器A分配到172.17.0.3，容器B分配到0.2
在这种情况下，两个容器的服务都会失效（容器A web服务访问提示ERR_HTTP_RESPONSE_CODE_FAILURE））
问题二：尝试不使用默认网络bridge，自建网络my_net，定义网段172.20.0.0/16，使用固定IP启动容器，在宿主机重启时并没有问题，但是这种方法容器不走宿主机ssr+代理（默认网桥绑定了docker0接口，在ssr+勾选即可，自己创建的并没有绑定接口）

期望的解决方案：容器在宿主机重启时正常启动并且走ssr+代理。

### (2) 路由器型号和固件版本

A：树莓派4B ImmortalWrt 18.06-5.4-SNAPSHOT r11579-375f97f0e5 (2022-01-25) / LuCI openwrt-18.06-k5.4 branch (git-22.021.28427-2a43828)

### (3) 详细日志

A：系统日志有一些我自己其他的操作，挑了一段刚开机的时候docker的日志，有需要我再在回复提供刚开机的时候的完整日志
Thu Feb  3 18:30:11 2022 kern.info kernel: [10598.974522] docker0: port 2(veth275c934) entered disabled state
Thu Feb  3 18:30:11 2022 kern.info kernel: [10598.980682] vetha501471: renamed from eth0
Thu Feb  3 18:30:11 2022 daemon.debug ModemManager[13285]: hotplug: event reported: action=add, name=vetha501471, subsystem=net
Thu Feb  3 18:30:11 2022 daemon.notice ttyd[12730]: rops_handle_POLLIN_netlink: DELADDR
Thu Feb  3 18:30:11 2022 kern.info kernel: [10599.150706] docker0: port 2(veth275c934) entered disabled state
Thu Feb  3 18:30:11 2022 daemon.err ModemManager[13285]: hotplug: parent device sysfspath not found
Thu Feb  3 18:30:11 2022 daemon.warn [4413]: <warn>  [vetha501471] invalid sysfs path read for net/vetha501471
Thu Feb  3 18:30:11 2022 daemon.warn [4413]: <warn>  [base-manager] couldn't handle kernel event: device net/vetha501471 not found
Thu Feb  3 18:30:11 2022 kern.info kernel: [10599.160567] device veth275c934 left promiscuous mode
Thu Feb  3 18:30:11 2022 kern.info kernel: [10599.165542] docker0: port 2(veth275c934) entered disabled state
Thu Feb  3 18:30:11 2022 daemon.debug ModemManager[13351]: hotplug: event reported: action=move, name=vetha501471, subsystem=net
Thu Feb  3 18:30:11 2022 daemon.warn [4413]: <warn>  [base-manager] couldn't handle kernel event: Invalid 'action' parameter given: 'move' (expected 'add' or 'remove')
Thu Feb  3 18:30:11 2022 daemon.debug ModemManager[13360]: hotplug: event reported: action=remove, name=vetha501471, subsystem=net
Thu Feb  3 18:30:11 2022 daemon.debug ModemManager[13375]: hotplug: event reported: action=remove, name=veth275c934, subsystem=net
Thu Feb  3 18:30:19 2022 kern.info kernel: [10606.936509] veth76bcd10: renamed from eth0
Thu Feb  3 18:30:19 2022 kern.info kernel: [10606.977830] docker0: port 1(vethf5b5691) entered disabled state
Thu Feb  3 18:30:19 2022 daemon.notice ttyd[12730]: rops_handle_POLLIN_netlink: DELADDR
Thu Feb  3 18:30:19 2022 kern.info kernel: [10607.051377] docker0: port 1(vethf5b5691) entered disabled state
Thu Feb  3 18:30:19 2022 kern.info kernel: [10607.060880] device vethf5b5691 left promiscuous mode
Thu Feb  3 18:30:19 2022 kern.info kernel: [10607.065859] docker0: port 1(vethf5b5691) entered disabled state
Thu Feb  3 18:30:19 2022 daemon.debug ModemManager[15762]: hotplug: event reported: action=add, name=veth76bcd10, subsystem=net
Thu Feb  3 18:30:19 2022 daemon.notice netifd: bridge 'docker0' link is down
Thu Feb  3 18:30:19 2022 daemon.notice netifd: Interface 'docker' has link connectivity loss
Thu Feb  3 18:30:19 2022 daemon.notice netifd: Interface 'docker1' has link connectivity loss
Thu Feb  3 18:30:19 2022 daemon.err ModemManager[15762]: hotplug: parent device sysfspath not found
Thu Feb  3 18:30:19 2022 daemon.warn [4413]: <warn>  [veth76bcd10] invalid sysfs path read for net/veth76bcd10
Thu Feb  3 18:30:19 2022 daemon.warn [4413]: <warn>  [base-manager] couldn't handle kernel event: device net/veth76bcd10 not found
Thu Feb  3 18:30:19 2022 daemon.debug ModemManager[15832]: hotplug: event reported: action=move, name=veth76bcd10, subsystem=net
Thu Feb  3 18:30:19 2022 daemon.warn [4413]: <warn>  [base-manager] couldn't handle kernel event: Invalid 'action' parameter given: 'move' (expected 'add' or 'remove')
Thu Feb  3 18:30:19 2022 daemon.debug ModemManager[15849]: hotplug: event reported: action=remove, name=veth76bcd10, subsystem=net
Thu Feb  3 18:30:19 2022 daemon.debug ModemManager[15874]: hotplug: event reported: action=remove, name=vethf5b5691, subsystem=net
Thu Feb  3 18:30:29 2022 daemon.err dockerd[9697]: time="2022-02-03T10:30:29.255365964Z" level=error msg="copy shim log" error="read /proc/self/fd/16: file already closed"
Thu Feb  3 18:30:29 2022 kern.info kernel: [10616.561673] veth6c2dd94: renamed from eth1
Thu Feb  3 18:30:29 2022 kern.info kernel: [10616.648052] docker1: port 3(vethfa4c37b) entered disabled state
Thu Feb  3 18:30:29 2022 daemon.notice ttyd[12730]: rops_handle_POLLIN_netlink: DELADDR
Thu Feb  3 18:30:29 2022 kern.info kernel: [10616.755746] docker1: port 3(vethfa4c37b) entered disabled state
Thu Feb  3 18:30:29 2022 kern.info kernel: [10616.768066] device vethfa4c37b left promiscuous mode
Thu Feb  3 18:30:29 2022 kern.info kernel: [10616.773050] docker1: port 3(vethfa4c37b) entered disabled state
Thu Feb  3 18:30:29 2022 daemon.debug ModemManager[21729]: hotplug: event reported: action=add, name=veth6c2dd94, subsystem=net
Thu Feb  3 18:30:29 2022 daemon.err ModemManager[21729]: hotplug: parent device sysfspath not found
Thu Feb  3 18:30:29 2022 daemon.warn [4413]: <warn>  [veth6c2dd94] invalid sysfs path read for net/veth6c2dd94
Thu Feb  3 18:30:29 2022 daemon.warn [4413]: <warn>  [base-manager] couldn't handle kernel event: device net/veth6c2dd94 not found
Thu Feb  3 18:30:29 2022 daemon.debug ModemManager[21805]: hotplug: event reported: action=move, name=veth6c2dd94, subsystem=net
Thu Feb  3 18:30:29 2022 daemon.warn [4413]: <warn>  [base-manager] couldn't handle kernel event: Invalid 'action' parameter given: 'move' (expected 'add' or 'remove')
Thu Feb  3 18:30:29 2022 daemon.debug ModemManager[21831]: hotplug: event reported: action=remove, name=veth6c2dd94, subsystem=net
Thu Feb  3 18:30:29 2022 daemon.debug ModemManager[21858]: hotplug: event reported: action=remove, name=vethfa4c37b, subsystem=net
Thu Feb  3 18:30:30 2022 daemon.err netdata[12051]: DISKSPACE: failed to statvfs() mount point '/mnt/sda2/docker/overlay2/7e99f63cd1f68315038ce75c134e25cd93300fa0245b2890de92c4244662494e/merged' (disk '_mnt_sda2_docker_overlay2_7e99f63cd1f68315038ce75c134e25cd93300fa0245b2890de92c4244662494e_merged', filesystem 'overlay', root '/')
Thu Feb  3 18:30:32 2022 daemon.err modprobe: failed to find a module named aufs

starxu1971, 2022-02-06:
这个是正常的，你要固定容器ip就要用虚拟网桥模式。

coolsnowwolf, 2026-09-06:
Docker 自建网络与 SSR+ 代理涉及自定义 bridge、路由和防火墙规则，缺少当前配置与抓包，关闭为“不计划处理”。

closed 2026-09-06 by coolsnowwolf