coolsnowwolf/lede #8827: ssr plus+插件绕过大陆IP模式会出现被回国模式名单影响的现象

smdx, 2022-02-05:
各位新年好，在ssr plus+插件选择绕过大陆IP模式会出现被回国模式的oversea_list.conf名单影响的现象，前几天自用的vps被封了443端口后哔哩哔哩、今日头条都打不开才发现的这个问题。
运行环境：
PVE 7.0-11、LXC 容器运行OpenWrt x86_64 Compiled by SometimesNa1ve, 2022.01.01 / LuCI Master (git-21.335.48743-5f363d9)
内核 5.11.22-4-pve，smartdns 国内走6053端口解析、国外走5335端口解析。

ssrplus+版本
[image]

相关设置：
[image]

[image]

[image]
问题描述：
遇到问题后试过更换GFW列表数据库、国内IP数据库都是一样访问不了，然后更改vps的端口正常翻墙后网站可以访问了就是播放地区视频会提示版权问题无法播放。把ssrplus+关闭后试用passwall播放视频没有提示版权问题了把vps关机后再试网站也正常访问。搜索issues后才知道有回国模式这个oversea_list.conf列表我进路由查看列表发现有相关域名走5335端口解析接着把列表文件删除后再转回用ssrplus+，相关网站都能正常访问了播放视频也没版权提示了再把vps关机测试国内域名访问都正常。很奇怪绕过大陆IP模式怎么会被回国模式的列表影响？不知道有没有遇到相同问题的朋友？谢谢

deplives, 2022-02-05:
所以这个和lede有啥关系？不去隔壁提？

smdx, 2022-02-05:
哎....我以为ssrplus+是作者的作品.....可以指个路吗？去反馈一下。谢谢

WYC-2020, 2022-02-06:
git搜索helloword

WYC-2020, 2022-02-06:
自己加的源居然不知道地址

smdx, 2022-02-06:
哎，固件是下载别人的不懂编译固件..谢谢了。我过去反馈一下。

closed 2022-02-12 by coolsnowwolf