coolsnowwolf/lede #8829: Patch failed!  Please fix openwrt/target/linux/bcm27xx/patches-5.4/950-0316-pinctrl-bcm2835-Add-support-for-BCM2711-pull-up-func.patch!

helmiau, 2022-02-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：Error compile

### (2) 路由器型号和固件版本

A：Raspberry Pi 4B, OpenWrt R22.1.1, Kernel 5.4.175

### (3) 详细日志

A：I use Github actions to compile firmware, then error occurs with logs below 

```
Applying /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/target/linux/bcm27xx/patches-5.4/950-0316-pinctrl-bcm2835-Add-support-for-BCM2711-pull-up-func.patch using plaintext: 
patching file drivers/pinctrl/bcm/pinctrl-bcm2835.c
Hunk #1 FAILED at 1168.
1 out of 1 hunk FAILED -- saving rejects to file drivers/pinctrl/bcm/pinctrl-bcm2835.c.rej
Patch failed!  Please fix /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/target/linux/bcm27xx/patches-5.4/950-0316-pinctrl-bcm2835-Add-support-for-BCM2711-pull-up-func.patch!
make[3]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.4.175/.prepared_e7203f9adae71e6c32d0e07494308c66] Error 1
Makefile:34: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.4.175/.prepared_e7203f9adae71e6c32d0e07494308c66' failed
make[3]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/target/linux/bcm27xx'
Makefile:11: recipe for target 'compile' failed
make[2]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/target/linux'
time: target/linux/compile#12.85#6.78#16.41
make[2]: *** [compile] Error 2
    ERROR: target/linux failed to build.
make[1]: *** [target/linux/compile] Error 1
target/Makefile:23: recipe for target 'target/linux/compile' failed
make: *** [target/compile] Error 2
make[1]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt'
```

keeno1982, 2022-02-07:
我也是这个问题，解决了吗？

EnnawYang, 2022-02-07:
> 我也是这个问题，解决了吗？

没有，等大佬们修吧

WYC-2020, 2022-02-07:
这玩意自己修复一下就可以了

WYC-2020, 2022-02-07:
[950-0316-pinctrl-bcm2835-Add-support-for-BCM2711-pull-up-func.zip](https://github.com/coolsnowwolf/lede/files/8013343/950-0316-pinctrl-bcm2835-Add-support-for-BCM2711-pull-up-func.zip)
吧这个解压 替换相同的patch 试下，因为我不编译这个型号没法测试，你们拿去测试一下
@EnnawYang @keeno1982

EnnawYang, 2022-02-07:
> [950-0316-pinctrl-bcm2835-Add-support-for-BCM2711-pull-up-func.zip](https://github.com/coolsnowwolf/lede/files/8013343/950-0316-pinctrl-bcm2835-Add-support-for-BCM2711-pull-up-func.zip) 吧这个解压 替换相同的patch 试下，因为我不编译这个型号没法测试，你们拿去测试一下 @EnnawYang @keeno1982

~~不瞒您说，我的树莓派在吃灰~~ ，没人试的话我试下

helmiau, 2022-02-07:
> 我也是这个问题，解决了吗？

No, this problem is still occured on me

helmiau, 2022-02-07:
> [950-0316-pinctrl-bcm2835-Add-support-for-BCM2711-pull-up-func.zip](https://github.com/coolsnowwolf/lede/files/8013343/950-0316-pinctrl-bcm2835-Add-support-for-BCM2711-pull-up-func.zip) 吧这个解压 替换相同的patch 试下，因为我不编译这个型号没法测试，你们拿去测试一下 @EnnawYang @keeno1982

Okay, i will compare this patch with latest lede patch then rebuild the firmware

helmiau, 2022-02-07:
> [950-0316-pinctrl-bcm2835-Add-support-for-BCM2711-pull-up-func.zip](https://github.com/coolsnowwolf/lede/files/8013343/950-0316-pinctrl-bcm2835-Add-support-for-BCM2711-pull-up-func.zip) 吧这个解压 替换相同的patch 试下，因为我不编译这个型号没法测试，你们拿去测试一下 @EnnawYang @keeno1982

i tried to build the firmware with your file, then this error occured

```
2022-02-07T13:21:08.6298405Z Applying /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/target/linux/bcm27xx/patches-5.4/950-0328-Revert-pinctrl-bcm2835-Pass-irqchip-when-adding-gpio.patch using plaintext: 
2022-02-07T13:21:08.6311010Z patching file drivers/pinctrl/bcm/pinctrl-bcm2835.c
2022-02-07T13:21:08.6314106Z Hunk #1 FAILED at 78.
2022-02-07T13:21:08.6316858Z Hunk #2 succeeded at 406 (offset 25 lines).
2022-02-07T13:21:08.6319959Z Hunk #3 succeeded at 1184 with fuzz 2 (offset 98 lines).
2022-02-07T13:21:08.6322630Z Hunk #4 FAILED at 1134.
2022-02-07T13:21:08.6326688Z 2 out of 4 hunks FAILED -- saving rejects to file drivers/pinctrl/bcm/pinctrl-bcm2835.c.rej
2022-02-07T13:21:08.6331242Z Patch failed!  Please fix /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/target/linux/bcm27xx/patches-5.4/950-0328-Revert-pinctrl-bcm2835-Pass-irqchip-when-adding-gpio.patch!
2022-02-07T13:21:08.6334669Z make[3]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.4.175/.prepared_0f37ba8fc107969211a8ee5770e4b1de] Error 1
2022-02-07T13:21:08.6338018Z make[2]: *** [compile] Error 2
2022-02-07T13:21:08.6338766Z Makefile:34: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.4.175/.prepared_0f37ba8fc107969211a8ee5770e4b1de' failed
2022-02-07T13:21:08.6346277Z make[3]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/target/linux/bcm27xx'
2022-02-07T13:21:08.6347033Z Makefile:11: recipe for target 'compile' failed
2022-02-07T13:21:08.6347854Z make[2]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/target/linux'
2022-02-07T13:21:08.6348384Z time: target/linux/compile#11.25#6.03#15.46
2022-02-07T13:21:08.6350091Z     ERROR: target/linux failed to build.
2022-02-07T13:21:08.6379914Z make[1]: *** [target/linux/compile] Error 1
2022-02-07T13:21:08.6380373Z target/Makefile:23: recipe for target 'target/linux/compile' failed
2022-02-07T13:21:08.6381123Z make: *** [target/compile] Error 2
2022-02-07T13:21:08.6383498Z make[1]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt'
2022-02-07T13:21:08.6384197Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/include/toplevel.mk:228: recipe for target 'target/compile' failed
2022-02-07T13:21:08.6437233Z ##[error]Process completed with exit code 2.
```

WYC-2020, 2022-02-07:
看来还有另外patch还要更新

WYC-2020, 2022-02-08:
@helmiau 看了这个补丁 现在5.4.175  相对于172 变化很大，很多补丁都已经不适合了，需要更新，我不敢随便搞了，这个变化太多 很多逻辑都变了，你要么等大佬更新，要么就回滚内核到172吧，我看172应该没问题，你或者可以一个个试 172-175 之间的版本 ，看你个人了

WYC-2020, 2022-02-08:
还有一个办法就是切换内核到5.10，我看官方op 这个bcm27xx就已经切换5.10了。应该不会再维护这个型号的5.4.175的补丁了
@helmiau

WYC-2020, 2022-02-08:
@helmiau @EnnawYang @keeno1982 试下这个提交，我这边自己编译没问题，功能的话 需要你们自己测试了
https://github.com/WYC-2020/lede/commit/6af5667c3db7c415d9c379bd05a1c96eee4477a5

helmiau, 2022-02-08:
@WYC-2020 your patches seems work here, I will tell you later when the firmware completely compiled.
Thank's for fix bro!.

[image]

helmiau, 2022-02-08:
@WYC-2020 your patch worked succesfully [WYC-2020/lede@6af5667](https://github.com/WYC-2020/lede/commit/6af5667c3db7c415d9c379bd05a1c96eee4477a5), may you can merge your commit to lede main branch. thank you.

WYC-2020, 2022-02-09:
@helmiau I just did it again, I personally feel that it is reasonable to delete these three patches 95-0316 950-0362 950-0328

WYC-2020, 2022-02-09:
@helmiau You can test it, after deleting these three patches

helmiau, 2022-02-09:
@WYC-2020 my firmware compiled successfully, thank's. Maybe you can make a pull request (if it necessary) with this commit https://github.com/WYC-2020/lede/commit/29017a5b3cec9b6768f0e01371a892e19ad97dc6
[image]

closed 2022-02-09 by coolsnowwolf