coolsnowwolf/lede #8833: 编译k2路由器固件时出错，请教大神如何解决，谢谢！

boeu1688, 2022-02-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：Makefile 'package/feeds/telephony/yate/Makefile' has a build dependency on 'fakeuname/host', which does not exist

### (2) 路由器型号和固件版本

A：路由器是k2，固件版本是PSG1208

### (3) 详细日志

A：
sed -i "s/Installed-Time: .*/Installed-Time: 1644066526/" /home/openwrt/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/status
rm -rf /home/openwrt/lede/build_dir/target-mipsel_24kc_musl/root-ramips/boot /home/openwrt/lede/build_dir/target-mipsel_24kc_musl/root-ramips/tmp/* /home/openwrt/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/info/*.postinst* /home/openwrt/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/lists/* /home/openwrt/lede/build_dir/target-mipsel_24kc_musl/root-ramips/var/lock/*.lock
find /home/openwrt/lede/build_dir/target-mipsel_24kc_musl/root-ramips/ -mindepth 1 -execdir touch -hcd "@1644066526" "{}" +
find: The relative path 'Files/WindowsApps/CanonicalGroupLimited.Ubuntu18.04onWindows_1804.2020.824.0_x64__79rhkp1fndgsc' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find.  Please remove that entry from $PATH
package/Makefile:66: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 1
make[2]: Leaving directory '/home/openwrt/lede'
package/Makefile:111: recipe for target '/home/openwrt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
make[1]: *** [/home/openwrt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/openwrt/lede'
/home/openwrt/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

slipdaly, 2022-02-07:
我的也出问题了，估计被谁提的代码搞坏了

WYC-2020, 2022-02-07:
你这显然看起来是path问题呢，并且看你这个应该不是ubuntu编译把，应该是win10那个自带模拟器编译的吧，应该和代码没关系

boeu1688, 2022-02-07:
> 我用的是win10自带的Linux子系统，是不是必须要用虚拟机或者实体机安装Ubuntu系统才行？

WYC-2020, 2022-02-07:
@boeu1688  这就不清楚了我从来不用win10自带的子系统 我都是虚拟机安装ubuntu 来编译的，这个问题你只能等用这个子系统编译的人给你说了 ，我记得以前有一个子系统编译的问题也是这个path问题

boeu1688, 2022-02-07:
> 

好勒，我去整个虚拟机安装Ubuntu再编译试试，有问题再来请教

xueJLcat, 2022-02-09:
我使用了ubuntu20.04.3和azure云20.04.3编都出现同样报错
若已解决请问如何解决相同报错。
谢谢

WYC-2020, 2022-02-10:
有时间看下吧，因为我都是编译x86 其他的型号很少编译

xueJLcat, 2022-02-10:
我编译的是树莓派4b的。同报这个错

xueJLcat, 2022-02-10:
官方仓库编译成功。lean的报题主类似的错误

AdoShan, 2022-02-20:
> @boeu1688 这就不清楚了我从来不用win10自带的子系统 我都是虚拟机安装ubuntu 来编译的，这个问题你只能等用这个子系统编译的人给你说了 ，我记得以前有一个子系统编译的问题也是这个path问题

大佬，我[这个](https://github.com/AdoShan/Build_OpenWrt/runs/5262733726?check_suite_focus=true#step:11:11923)错误能帮忙看看啥原因吗，每次都是卡在同一个地方。编译固件的第11923行开始报的错误
`make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.`

anjue39, 2022-03-10:
> > @boeu1688 这就不清楚了我从来不用win10自带的子系统 我都是虚拟机安装ubuntu 来编译的，这个问题你只能等用这个子系统编译的人给你说了 ，我记得以前有一个子系统编译的问题也是这个path问题
> 
> 大佬，我[这个](https://github.com/AdoShan/Build_OpenWrt/runs/5262733726?check_suite_focus=true#step:11:11923)错误能帮忙看看啥原因吗，每次都是卡在同一个地方。我使用Ubuntu20.04.3LST生成并上传了.config文件并编译，编译固件的第11923行开始报的错误 `make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2 Error: Process completed with exit code 2.`

我也经常遇到这个问题

coolsnowwolf, 2026-09-05:
错误来自旧 telephony/yate 依赖 `fakeuname/host`，当前 feeds 已不存在该 yate 包，且日志还包含 Windows PATH 安全错误，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf