coolsnowwolf/lede #8836: 同步最新源码出现这样的提示 “WARNING: Makefile 'package/network/services/uhttpd/Makefile' has a dependency on 'libucode', which does not exist”

mitsukileung, 2022-02-08:
X86_64
同步最新源码出现这样的提示 “WARNING: Makefile 'package/network/services/uhttpd/Makefile' has a dependency on 'libucode', which does not exist”

WYC-2020, 2022-02-08:
增加这个库https://github.com/openwrt/openwrt/tree/master/package/utils/ucode。提交不完整

mitsukileung, 2022-02-08:
> 增加这个库[https://github.com/openwrt/openwrt/tree/master/package/utils/ucode。提交不完整](https://github.com/openwrt/openwrt/tree/master/package/utils/ucode%E3%80%82%E6%8F%90%E4%BA%A4%E4%B8%8D%E5%AE%8C%E6%95%B4)

谢谢大佬，修改了makefile后正常了

lifetyper, 2022-02-08:
还是等大佬把这个包导进来吧

rongcenter, 2022-02-08:
同

Beginner-Go, 2022-02-08:
晚点我来弄

Cptmacmillan2022007, 2022-02-08:
我今天在线编译也出现了这个情况

helmiau, 2022-02-08:
@Cptmacmillan2022007 your logs same is mine, i still waiting for fix from other dev

```
2022-02-08T12:26:03.2485635Z FAILED: CMakeFiles/uhttpd.dir/main.c.o 
2022-02-08T12:26:03.2488548Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/bin/aarch64-openwrt-linux-musl-gcc -DHAVE_LUA -DHAVE_SHADOW -DHAVE_TLS -DHAVE_UBUS -DHAVE_UCODE -D_FILE_OFFSET_BITS=64  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-aarch64_cortex-a72_musl/uhttpd-2022-02-07-2f8b1360=uhttpd-2022-02-07-2f8b1360 -Wformat -Werror=format-security -DPIC -fPIC -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG   -Os -Wall -Werror -Wmissing-declarations --std=gnu99 -g3 -MD -MT CMakeFiles/uhttpd.dir/main.c.o -MF CMakeFiles/uhttpd.dir/main.c.o.d -o CMakeFiles/uhttpd.dir/main.c.o -c /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-aarch64_cortex-a72_musl/uhttpd-2022-02-07-2f8b1360/main.c
2022-02-08T12:26:03.2490710Z make[2]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-aarch64_cortex-a72_musl/uhttpd-2022-02-07-2f8b1360/.built] Error 1
2022-02-08T12:26:03.2498260Z In file included from /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-aarch64_cortex-a72_musl/uhttpd-2022-02-07-2f8b1360/main.c:39:
2022-02-08T12:26:03.2499172Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-aarch64_cortex-a72_musl/uhttpd-2022-02-07-2f8b1360/uhttpd.h:37:10: fatal error: ucode/vm.h: No such file or directory
2022-02-08T12:26:03.2499664Z  #include <ucode/vm.h>
2022-02-08T12:26:03.2499927Z           ^~~~~~~~~~~~
2022-02-08T12:26:03.2500200Z compilation terminated.
2022-02-08T12:26:03.2500515Z ninja: build stopped: subcommand failed.
2022-02-08T12:26:03.2501239Z Makefile:135: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-aarch64_cortex-a72_musl/uhttpd-2022-02-07-2f8b1360/.built' failed
2022-02-08T12:26:03.2501938Z make[2]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/package/network/services/uhttpd'
2022-02-08T12:26:03.2502395Z time: package/network/services/uhttpd/compile#0.20#0.07#0.26
2022-02-08T12:26:03.2502765Z     ERROR: package/network/services/uhttpd failed to build.
2022-02-08T12:26:03.2544378Z make[1]: *** [package/network/services/uhttpd/compile] Error 1
2022-02-08T12:26:03.2544937Z package/Makefile:114: recipe for target 'package/network/services/uhttpd/compile' failed
2022-02-08T12:26:03.2549597Z make: *** [package/compile] Error 2
2022-02-08T12:26:03.2550110Z make[1]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt'
2022-02-08T12:26:03.2551099Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/include/toplevel.mk:228: recipe for target 'package/compile' failed
2022-02-08T12:26:03.2595050Z ##[error]Process completed with exit code 2.
```

wokilp, 2022-02-08:
还以为就我遇到这个问题

dysl357085918, 2022-02-08:
same error

HollyMadCow, 2022-02-08:
用大佬的makefile替换 package/utils/ucode/下面的makefile就可以了，我自己摸索着解决了，到这里一看，原来已经发现了啊，我原来是去https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=package/utils/ucode/Makefile;h=4d66cc9f6c91263fd468a00b839ad5381c3740cb;hb=HEAD这里下载这个makefile替换掉就可以编译的

helmiau, 2022-02-08:
@Beginner-Go waiting for merge then. thanks.

closed 2022-02-09 by AmadeusGhost