coolsnowwolf/lede #8837: 2月8日源码更新后无法编译

Zaphkito, 2022-02-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：今天更新源码后无法编译完成，编译之前的同样设置可以编译成功（本地和github action编译都有这问题，而且github action编译前后没有更改代码）

### (2) 路由器型号和固件版本

A：X86_64

### (3) 详细日志

A：SHELL= flock /home/zaphkito/Source/OpenWrt/tmp/.root-copy.flock -c 'cp -fpR /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/ruby-3.0.3/.pkgdir/ruby-strscan/. /home/zaphkito/Source/OpenWrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/zaphkito/Source/OpenWrt/staging_dir/target-x86_64_musl/root-x86/stamp/.ruby-strscan_installed
echo "ruby-strscan" >> /home/zaphkito/Source/OpenWrt/staging_dir/target-x86_64_musl/pkginfo/ruby.default.install
mkdir -p /home/zaphkito/Source/OpenWrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/zaphkito/Source/OpenWrt/tmp/.root-copy.flock -c 'cp -fpR /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/ruby-3.0.3/.pkgdir/ruby-yaml/. /home/zaphkito/Source/OpenWrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/zaphkito/Source/OpenWrt/staging_dir/target-x86_64_musl/root-x86/stamp/.ruby-yaml_installed
echo "ruby-yaml" >> /home/zaphkito/Source/OpenWrt/staging_dir/target-x86_64_musl/pkginfo/ruby.default.install
make[3]: Leaving directory '/home/zaphkito/Source/OpenWrt/feeds/packages/lang/ruby'
time: package/feeds/packages/ruby/compile#209.16#28.83#236.54
make[2]: Leaving directory '/home/zaphkito/Source/OpenWrt'
make[1]: *** [package/Makefile:110: /home/zaphkito/Source/OpenWrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zaphkito/Source/OpenWrt'
make: *** [/home/zaphkito/Source/OpenWrt/include/toplevel.mk:230：world] 错误 2

[image]
[image]

Beginner-Go, 2022-02-08:
我看看

Karmylr, 2022-02-08:
+`

hh1100, 2022-02-08:
我用的GitHub action 在complie这个出错 +1 ，看来出错的人很多呀

WYC-2020, 2022-02-08:
我的没出错 只是我编译的是x86

GreatMichaelLee, 2022-02-08:
按 #8836 里面把ucode Makefile修复了就好了

closed 2022-02-09 by AmadeusGhost

hh1100, 2022-02-11:
已经可以正常编译了