coolsnowwolf/lede #8848: 5.15编译报错

Karmylr, 2022-02-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译了几次都是一模一样的报错

### (2) 路由器型号和固件版本

A：

### (3) 详细日志
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency antfs-mount for automount
 * pkg_hash_fetch_best_installation_candidate: Packages for automount found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package automount.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/conter/lede'
make[1]: *** [package/Makefile:111: /home/conter/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/conter/lede'
make: *** [/home/conter/lede/include/toplevel.mk:230: world] Error 2

A：
[image]

WYC-2020, 2022-02-09:
取消编译antfs 那个是非5.15的包

Karmylr, 2022-02-09:
> 取消编译antfs 那个是非5.15的包

applications里面找不到呢 是在哪里呢

WYC-2020, 2022-02-09:
@Karmylr config 文件里搜索antfs-mount  吧他改成不编译即可，你选5.15这个包在menuconfig里面是不会出来的

Karmylr, 2022-02-09:
> 

[image]
请问怎么改呢

WYC-2020, 2022-02-09:
@Karmylr  看你这个配置文件是没有编译antfs的 你是不是先编译过非5.15 然后你有切换会5.15 接到编译的，这样的话 删除build目录里面的automount文件夹

Karmylr, 2022-02-09:
@WYC-2020 多谢！确实是先编译了5.10的。现在已经解决了

closed 2022-02-09 by Karmylr

WYC-2020, 2022-02-09:
这玩意其实也算一个bug，因为编译过其他的版本后 automount 记录了当前内核所依赖东西，当你切换后，但是automount并没有变化，所以他不会更新，导致在最后会失败，看makefile又没什么方法让其每次都更新，这个等大佬看吧