coolsnowwolf/lede #8857: sftp报128错误是什么原因？

GreatMichaelLee, 2022-02-10:
如题，已选vsftp 和Luci-app vsftp 以及Network->SSH->勾选了openssh-sftp-server,为何用winscp连接为sftp模式的时候就报下面的128错误呢？用scp没问题

[image]

GreatMichaelLee, 2022-02-11:
发现这个问题是跟#8260相关，我自己编译出来的x86固件里openssh-sftp版本是8.8p1-2,是不行的,出现上述错误，但是手工安装 https://mirrors.cloud.tencent.com/lede/snapshots/packages/x86_64/packages/openssh-sftp-server_8.8p1-3_x86_64.ipk
这个lede的源里的8.8p1-3的版本就work了。

同样的我在R2S上编译出来的版本却是8.8p1-1, 也是work的，我就奇了怪了，makefile里都是一样的，咋就一个出8.8p1-1一个是8.8p1-2而且 1/3版本OK,  2不行呢？求大佬解惑。

GreatMichaelLee, 2022-02-13:
发现源上更新到8.8p1-5, 也很好用，但是代码编译怎么用到这个不清楚，哪位大佬看看？

https://mirrors.cloud.tencent.com/lede/snapshots/packages/x86_64/packages/openssh-sftp-server_8.8p1-5_x86_64.ipk

GreatMichaelLee, 2022-04-16:
真棒，发现今天更新的openssh 9.0p1 解决了这个问题，可以close了 @coolsnowwolf

WYC-2020, 2022-04-16:
> 真棒，发现今天更新的openssh-sftp 9.0p1 解决了这个问题，可以close了 @coolsnowwolf

你确定哈哈，反正从8.8以后 我验证的所有都不行

coolsnowwolf, 2026-09-05:
SFTP 128 错误只有客户端截图，没有 sshd/sftp 子系统日志和实际配置；无法确认是 openssh-sftp-server、权限还是 WinSCP 参数，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf