coolsnowwolf/lede #8862: 最新代码报这个错，什么原因？

GreatMichaelLee, 2022-02-11:
make[3]: Entering directory '/home/builder/lede_x86/feeds/packages/libs/boost'
Selected Boost API sysv for architecture x86_64 and cpu type " "
mkdir -p /home/builder/lede_x86/dl
SHELL= flock /home/builder/lede_x86/tmp/.boost_1_76_0.tar.bz2.flock -c '        /home/builder/lede_x86/scripts/download.pl "/home/builder/lede_x86/dl" "boost_1_76_0.tar.bz2" "f0397ba6e982c4450f27bf32a2a83292aba035b827a5623a14636ea583318c41" "" "@SF/boost/boost/1.76.0" "https://dl.bintray.com/boostorg/release/1.76.0/source/"    '
. /home/builder/lede_x86/include/shell.sh; bzcat /home/builder/lede_x86/dl/boost_1_76_0.tar.bz2 | tar -C /home/builder/lede_x86/build_dir/hostpkg/boost_1_76_0/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /home/builder/lede_x86/build_dir/hostpkg/boost_1_76_0

Applying ./patches/010-mips64-fiber.patch using plaintext:
patching file boost/fiber/detail/cpu_relax.hpp

Applying ./patches/020-math.patch using plaintext:
patching file boost/math/special_functions/detail/bernoulli_details.hpp
Hunk #1 FAILED at 360.
Hunk #2 FAILED at 382.
Hunk #3 FAILED at 464.
Hunk #4 FAILED at 494.
Hunk #5 FAILED at 555.
5 out of 5 hunks FAILED -- saving rejects to file boost/math/special_functions/detail/bernoulli_details.hpp.rej
patching file boost/math/tools/atomic.hpp
Hunk #1 succeeded at 18 with fuzz 1 (offset 2 lines).
patching file libs/math/test/Jamfile.v2
Hunk #1 succeeded at 122 (offset 1 line).
Hunk #2 FAILED at 877.
Hunk #3 succeeded at 1090 (offset -54 lines).
Hunk #4 FAILED at 1173.
2 out of 4 hunks FAILED -- saving rejects to file libs/math/test/Jamfile.v2.rej
patching file libs/math/test/bernoulli_no_atomic_d.cpp
patching file libs/math/test/bernoulli_no_atomic_fail.cpp
patching file libs/math/test/bernoulli_no_atomic_mp.cpp
Patch failed!  Please fix ./patches/020-math.patch!
make[3]: *** [Makefile:505: /home/builder/lede_x86/build_dir/hostpkg/boost_1_76_0/.preparedec24eaa8f708f9b7014d6ffbcdfa071a_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/libs/boost'
time: package/feeds/packages/boost/host-compile#17.07#24.90#33.03
    ERROR: package/feeds/packages/boost [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/boost/host/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：world] 错误 2
builder@Build-Server:~/lede_x86$

Beginner-Go, 2022-02-11:
@coolsnowwolf 为什么要回退boost呢？我这边x86编译没问题

Beginner-Go, 2022-02-11:
应该现在好了

Jason6111, 2022-02-11:
重新
git pull
./scripts/feeds update -a && ./scripts/feeds install -a

closed 2022-02-12 by coolsnowwolf