coolsnowwolf/lede #8869: smrtdns设置完成后用nslookup命令，显示DNS超时

zhengjun131461, 2022-02-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
OpenWrt R22.2.2 / LuCI Master (git-21.335.48743-5f363d9)
### (1) 具体问题
smrtdns设置完成后用nslookup命令，显示DNS超时 
A：

### (2) 路由器型号和固件版本
smrtdns设置完成后用nslookup命令，显示DNS超时 
A：
https://user-images.githubusercontent.com/43834414/152638833-6bfe8090-b5b5-4206-a4ad-90edca8f215a.png
但是在后面加入DNS服务就可以解析
https://user-images.githubusercontent.com/43834414/152638859-f4854d3a-7e73-4484-ba15-afdc7079dfcb.png
### (3) 详细日志

A：

zhengjun131461, 2022-02-13:
要在那个文件修改，学习一下

zhengjun131461, 2022-02-13:
https://github.com/pymumu/smartdns是这个地址，是不是这个源码

zhengjun131461, 2022-02-14:
URL不用改吗？发自我的手机-------- 原始邮件 --------发件人： michael lee ***@***.***>日期： 2022年2月13日周日 23:14收件人： coolsnowwolf/lede ***@***.***>抄送： zhengjun131461 ***@***.***>, Author ***@***.***>主    题： Re: [coolsnowwolf/lede] smrtdns设置完成后用nslookup命令，显示DNS超时  (Issue #8869)

https://github.com/pymumu/smartdns是这个地址，是不是这个源码

是的，修改feeds/packages/net/smartdns/Makefile,修改内容见图

zhengjun131461, 2022-02-14:
之前的包是不是可以删除啊

zhengjun131461, 2022-02-14:
[image]
今天编译的

naoki66, 2022-03-27:
root@OpenWrtUNICOM:~# iptables -t nat -L PREROUTING | grep REDIRECT
REDIRECT   udp  --  anywhere             anywhere             udp dpt:domain redir ports 53
REDIRECT   tcp  --  anywhere             anywhere             tcp dpt:domain redir ports 53

这是没成功？

coolsnowwolf, 2026-09-05:
smrtdns DNS 超时没有设备、监听端口、dnsmasq 配置或服务日志；加入额外 DNS 后正常说明是配置链路问题，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf