coolsnowwolf/lede #8874: kmod-drm missing dependencies: drm_panel_orientation_quirks.ko

helmiau, 2022-02-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：Package kmod-drm is missing dependencies for the following libraries: drm_panel_orientation_quirks.ko

### (2) 路由器型号和固件版本

A：Raspberry Pi 4B, OpenWrt R22.1.1, Kernel 5.4.175, lede source before https://github.com/coolsnowwolf/lede/commit/8f6139bf163c5dd741d7d0ab503b8560157e44bb commit

### (3) 详细日志

```
make[2]: Entering directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/package/kernel/linux'
Package kmod-drm is missing dependencies for the following libraries:
drm_panel_orientation_quirks.ko
modules/video.mk:262: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/bin/targets/bcm27xx/bcm2711/packages/kmod-drm_5.4.175-1_aarch64_cortex-a72.ipk' failed
make[2]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/package/kernel/linux'
make[2]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/bin/targets/bcm27xx/bcm2711/packages/kmod-drm_5.4.175-1_aarch64_cortex-a72.ipk] Error 1
time: package/kernel/linux/compile#3.46#0.43#3.87
    ERROR: package/kernel/linux failed to build.
package/Makefile:114: recipe for target 'package/kernel/linux/compile' failed
make[1]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt'
make[1]: *** [package/kernel/linux/compile] Error 1
make: *** [package/compile] Error 2
/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/include/toplevel.mk:228: recipe for target 'package/compile' failed
Error: Process completed with exit code 2.
```

WYC-2020, 2022-02-14:
@helmiau 
https://github.com/WYC-2020/lede/commit/471381c40a450009f3f29bb1b31f8f2b279e6a5e

helmiau, 2022-02-14:
@WYC-2020 ok i will test it, thank you

closed 2022-02-14 by coolsnowwolf

helmiau, 2022-02-14:
@WYC-2020 now error occured
```
make[2]: Entering directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/package/kernel/linux'
modules/video.mk:7: *** missing separator.  Stop.
```

WYC-2020, 2022-02-14:
@helmiau  I tested without any problem

WYC-2020, 2022-02-14:
@helmiau  send your video.mk file to have a look

helmiau, 2022-02-14:
i replaced video.mk file from your commit https://github.com/WYC-2020/lede/commit/471381c40a450009f3f29bb1b31f8f2b279e6a5e without changing anything inside.
Then error shown `make[2]: Entering directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/package/kernel/linux'
modules/video.mk:7: *** missing separator.  Stop.`

helmiau, 2022-02-14:
@WYC-2020 i tried to compile again, and error log stil same :
```
make[2]: Entering directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/package/kernel/linux' modules/video.mk:7: *** missing separator. Stop.
```

helmiau, 2022-02-14:
@WYC-2020 @coolsnowwolf i just re-read the logs, and i found the problem.. please take a look at this.

```
Package kmod-drm-vc4 is missing dependencies for the following libraries:
1669
fb_sys_fops.ko
1670
syscopyarea.ko
1671
sysfillrect.ko
1672
sysimgblt.ko

```

WYC-2020, 2022-02-15:
@helmiau You have a problem with another driver, I'll wait and see

WYC-2020, 2022-02-15:
@helmiau  https://github.com/WYC-2020/lede/commit/f3b81b3e65fdcfc1e533ee72377aa16063908cf7

helmiau, 2022-02-16:
@WYC-2020 thank you man, no error anymore around video.mk

helmiau, 2022-02-16:
> @helmiau [WYC-2020/lede@f3b81b3](https://github.com/WYC-2020/lede/commit/f3b81b3e65fdcfc1e533ee72377aa16063908cf7)

@WYC-2020 can you make a pull request to coolsnowwolf/lede master branch?