coolsnowwolf/lede #8875: feeds install 一堆warning

GreatMichaelLee, 2022-02-14:
最新commit, 报一堆warning，麻烦大佬看下

builder@Build-Server:~/lede_x86$ ./scripts/feeds install -a
WARNING: Makefile 'package/lean/luci-app-aliyundrive-webdav/Makefile' has a dependency on 'aliyundrive-webdav', which does not exist
WARNING: Makefile 'package/lean/luci-app-amule/Makefile' has a dependency on 'amule', which does not exist
WARNING: Makefile 'package/lean/luci-app-baidupcs-web/Makefile' has a dependency on 'baidupcs-web', which does not exist
WARNING: Makefile 'package/lean/luci-app-dnsforwarder/Makefile' has a dependency on 'dnsforwarder', which does not exist
WARNING: Makefile 'package/lean/luci-app-frpc/Makefile' has a dependency on 'frpc', which does not exist
WARNING: Makefile 'package/lean/luci-app-frps/Makefile' has a dependency on 'frps', which does not exist
WARNING: Makefile 'package/lean/luci-app-go-aliyundrive-webdav/Makefile' has a dependency on 'go-aliyundrive-webdav', which does not exist
WARNING: Makefile 'package/lean/luci-app-nps/Makefile' has a dependency on 'npc', which does not exist
WARNING: Makefile 'package/lean/luci-app-qbittorrent/Makefile' has a dependency on 'qBittorrent-static', which does not exist
WARNING: Makefile 'package/lean/luci-app-qbittorrent/Makefile' has a dependency on 'qbittorrent', which does not exist
WARNING: Makefile 'package/lean/luci-app-rclone/Makefile' has a dependency on 'rclone-config', which does not exist
WARNING: Makefile 'package/lean/luci-app-rclone/Makefile' has a dependency on 'rclone-webui-react', which does not exist
WARNING: Makefile 'package/lean/luci-app-rclone/Makefile' has a dependency on 'rclone-ng', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'redsocks2', which does not exist
WARNING: Makefile 'package/lean/luci-app-turboacc/Makefile' has a dependency on 'dnsforwarder', which does not exist
WARNING: Makefile 'package/lean/luci-app-unblockmusic/Makefile' has a dependency on 'UnblockNeteaseMusic-Go', which does not exist
WARNING: Makefile 'package/lean/luci-app-unblockmusic/Makefile' has a dependency on 'UnblockNeteaseMusic', which does not exist
WARNING: Makefile 'package/lean/luci-app-uugamebooster/Makefile' has a dependency on 'uugamebooster', which does not exist
WARNING: Makefile 'package/lean/luci-app-verysync/Makefile' has a dependency on 'verysync', which does not exist
WARNING: Makefile 'package/lean/luci-app-vlmcsd/Makefile' has a dependency on 'vlmcsd', which does not exist
Installing all packages from feed packages.
WARNING: Not overriding core package 'aliyundrive-webdav'; use -f to force
WARNING: Not overriding core package 'amule'; use -f to force
WARNING: Not overriding core package 'antileech'; use -f to force
WARNING: Not overriding core package 'baidupcs-web'; use -f to force
WARNING: Not overriding core package 'dns2socks'; use -f to force
WARNING: Not overriding core package 'dnsforwarder'; use -f to force
WARNING: Not overriding core package 'frp'; use -f to force
WARNING: Not overriding core package 'gmediarender'; use -f to force
WARNING: Not overriding core package 'go-aliyundrive-webdav'; use -f to force
WARNING: Not overriding core package 'ipt2socks'; use -f to force
WARNING: Not overriding core package 'microsocks'; use -f to force
WARNING: Not overriding core package 'nps'; use -f to force
WARNING: Not overriding core package 'pdnsd-alt'; use -f to force
WARNING: Not overriding core package 'qBittorrent'; use -f to force
WARNING: Not overriding core package 'qBittorrent-static'; use -f to force
WARNING: Not overriding core package 'qtbase'; use -f to force
WARNING: Not overriding core package 'qttools'; use -f to force
WARNING: Not overriding core package 'rblibtorrent'; use -f to force
WARNING: Not overriding core package 'rclone'; use -f to force
WARNING: Not overriding core package 'rclone-ng'; use -f to force
WARNING: Not overriding core package 'rclone-webui-react'; use -f to force
WARNING: Not overriding core package 'redsocks2'; use -f to force
WARNING: Not overriding core package 'tcpping'; use -f to force
WARNING: Not overriding core package 'UnblockNeteaseMusic'; use -f to force
WARNING: Not overriding core package 'UnblockNeteaseMusic-Go'; use -f to force
WARNING: Not overriding core package 'uugamebooster'; use -f to force
WARNING: Not overriding core package 'verysync'; use -f to force
WARNING: Not overriding core package 'vlmcsd'; use -f to force
WARNING: Not overriding core package 'wxbase'; use -f to force
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed helloworld.
WARNING: Not overriding core package 'simple-obfs'; use -f to force
WARNING: Not overriding core package 'trojan'; use -f to force

GreatMichaelLee, 2022-02-14:
And then build fail........................

```
make[7]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/src/rc'
Making all in test
make[7]: Entering directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/src/test'
x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../..   -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include   -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par=pdnsd-1.2.9b-par -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/builder/lede_x86/staging_dir/target-x86_64_musl/usr/include  -pthread -c ../icmp.c
In file included from ../icmp.c:31:
/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
 #warning redirecting incorrect #include <sys/poll.h> to <poll.h>
  ^~~~~~~
In file included from ../dns.h:30,
                 from ../cache.h:31,
                 from ../helpers.h:32,
                 from ../error.h:35,
                 from ../icmp.c:63:
/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/net/if.h:64:8: error: redefinition of 'struct ifmap'
 struct ifmap {
        ^~~~~
In file included from /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/linux/icmp.h:23,
                 from ../icmp.c:44:
/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/linux/if.h:194:8: note: originally defined here
 struct ifmap {
        ^~~~~
In file included from ../dns.h:30,
                 from ../cache.h:31,
                 from ../helpers.h:32,
                 from ../error.h:35,
                 from ../icmp.c:63:
/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/net/if.h:76:8: error: redefinition of 'struct ifreq'
 struct ifreq {
        ^~~~~
In file included from /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/linux/icmp.h:23,
                 from ../icmp.c:44:
/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/linux/if.h:232:8: note: originally defined here
 struct ifreq {
        ^~~~~
In file included from ../dns.h:30,
                 from ../cache.h:31,
                 from ../helpers.h:32,
                 from ../error.h:35,
                 from ../icmp.c:63:
/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/net/if.h:116:8: error: redefinition of 'struct ifconf'
 struct ifconf {
        ^~~~~~
In file included from /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/linux/icmp.h:23,
                 from ../icmp.c:44:
/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/linux/if.h:284:8: note: originally defined here
 struct ifconf  {
        ^~~~~~
make[7]: *** [Makefile:460: icmp.o] Error 1
make[7]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/src/test'
make[6]: *** [Makefile:607: all-recursive] Error 1
make[6]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/src'
Making all in doc
make[6]: Entering directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/doc'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/doc'
Making all in contrib
make[6]: Entering directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/contrib'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/contrib'
make[6]: Entering directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par'
make[6]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par'
make[5]: *** [Makefile:270: all-recursive] Error 1
make[5]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par'
make[4]: *** [Makefile:202: all] Error 2
make[4]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par'
make[3]: *** [Makefile:59: /home/builder/lede_x86/build_dir/target-x86_64_musl/pdnsd-1.2.9b-par/.built] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/package/lean/openwrt-passwall/pdnsd-alt'
time: package/lean/openwrt-passwall/pdnsd-alt/compile#0.26#0.12#0.34
    ERROR: package/lean/openwrt-passwall/pdnsd-alt failed to build.
make[2]: *** [package/Makefile:116: package/lean/openwrt-passwall/pdnsd-alt/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：world] 错误 2
builder@Build-Server:~/lede_x86$
```

WYC-2020, 2022-02-14:
哎，feed文件夹删除了重新更新源啊

GreatMichaelLee, 2022-02-14:
> 哎，feed文件夹删除了重新更新源啊

哎，我是菜鸟啊, 没有Q&A不知道这种trick啊 ^_^

GreatMichaelLee, 2022-02-14:
> 哎，feed文件夹删除了重新更新源啊

但是，哎，还是一样的啊...

```
builder@Build-Server:~/lede_r4s$ cd feeds
builder@Build-Server:~/lede_r4s/feeds$ ll
总用量 48
drwxrwxr-x 12 builder builder 4096 2月  14 09:21 ./
drwxrwxrwx 19 builder builder 4096 2月  14 09:21 ../
drwxrwxr-x 17 builder builder 4096 1月   1 23:01 helloworld/
lrwxrwxrwx  1 builder builder   27 2月  14 09:21 helloworld.index -> helloworld.tmp/.packageinfo
lrwxrwxrwx  1 builder builder   26 2月  14 09:21 helloworld.targetindex -> helloworld.tmp/.targetinfo
drwxrwxr-x  3 builder builder 4096 2月  14 09:21 helloworld.tmp/
drwxrwxr-x 12 builder builder 4096 8月  12  2021 luci/
lrwxrwxrwx  1 builder builder   21 2月  14 09:21 luci.index -> luci.tmp/.packageinfo
lrwxrwxrwx  1 builder builder   20 2月  14 09:21 luci.targetindex -> luci.tmp/.targetinfo
drwxrwxr-x  3 builder builder 4096 2月  14 09:21 luci.tmp/
drwxrwxr-x 17 builder builder 4096 8月  12  2021 packages/
lrwxrwxrwx  1 builder builder   25 2月  14 09:21 packages.index -> packages.tmp/.packageinfo
lrwxrwxrwx  1 builder builder   24 2月  14 09:21 packages.targetindex -> packages.tmp/.targetinfo
drwxrwxr-x  3 builder builder 4096 2月  14 09:21 packages.tmp/
drwxrwxr-x 37 builder builder 4096 8月  12  2021 routing/
lrwxrwxrwx  1 builder builder   24 2月  14 09:21 routing.index -> routing.tmp/.packageinfo
lrwxrwxrwx  1 builder builder   23 2月  14 09:21 routing.targetindex -> routing.tmp/.targetinfo
drwxrwxr-x  3 builder builder 4096 2月  14 09:21 routing.tmp/
drwxrwxr-x  6 builder builder 4096 8月  12  2021 telephony/
lrwxrwxrwx  1 builder builder   26 2月  14 09:21 telephony.index -> telephony.tmp/.packageinfo
lrwxrwxrwx  1 builder builder   25 2月  14 09:21 telephony.targetindex -> telephony.tmp/.targetinfo
drwxrwxr-x  3 builder builder 4096 2月  14 09:21 telephony.tmp/
builder@Build-Server:~/lede_r4s/feeds$ rm -rf *
builder@Build-Server:~/lede_r4s/feeds$ cd ..
builder@Build-Server:~/lede_r4s$ ./scripts/feeds update -a
Updating feed 'packages' from 'https://github.com/coolsnowwolf/packages' ...
正克隆到 './feeds/packages'...
remote: Enumerating objects: 6332, done.
remote: Counting objects: 100% (6332/6332), done.
remote: Compressing objects: 100% (5232/5232), done.
remote: Total 6332 (delta 234), reused 4258 (delta 181), pack-reused 0
接收对象中: 100% (6332/6332), 4.30 MiB | 3.88 MiB/s, 完成.
处理 delta 中: 100% (234/234), 完成.
Updating feed 'luci' from 'https://github.com/coolsnowwolf/luci' ...
正克隆到 './feeds/luci'...
remote: Enumerating objects: 3800, done.
remote: Counting objects: 100% (3800/3800), done.
remote: Compressing objects: 100% (1737/1737), done.
remote: Total 3800 (delta 826), reused 3557 (delta 771), pack-reused 0
接收对象中: 100% (3800/3800), 3.62 MiB | 13.59 MiB/s, 完成.
处理 delta 中: 100% (826/826), 完成.
Updating feed 'routing' from 'https://git.openwrt.org/feed/routing.git' ...
正克隆到 './feeds/routing'...
remote: Enumerating objects: 393, done.
remote: Counting objects: 100% (393/393), done.
remote: Compressing objects: 100% (321/321), done.
remote: Total 393 (delta 26), reused 280 (delta 17), pack-reused 0
接收对象中: 100% (393/393), 278.88 KiB | 806.00 KiB/s, 完成.
处理 delta 中: 100% (26/26), 完成.
Updating feed 'telephony' from 'https://git.openwrt.org/feed/telephony.git' ...
正克隆到 './feeds/telephony'...
remote: Enumerating objects: 283, done.
remote: Counting objects: 100% (283/283), done.
remote: Compressing objects: 100% (270/270), done.
remote: Total 283 (delta 7), reused 69 (delta 1), pack-reused 0
接收对象中: 100% (283/283), 190.33 KiB | 568.00 KiB/s, 完成.
处理 delta 中: 100% (7/7), 完成.
Updating feed 'helloworld' from 'https://github.com/fw876/helloworld.git' ...
正克隆到 './feeds/helloworld'...
remote: Enumerating objects: 162, done.
remote: Counting objects: 100% (162/162), done.
remote: Compressing objects: 100% (131/131), done.
[... 50 lines cut ...]
WARNING: Not overriding core package 'microsocks'; use -f to force
WARNING: Not overriding core package 'nps'; use -f to force
WARNING: Not overriding core package 'pdnsd-alt'; use -f to force
WARNING: Not overriding core package 'qBittorrent'; use -f to force
WARNING: Not overriding core package 'qBittorrent-static'; use -f to force
WARNING: Not overriding core package 'qtbase'; use -f to force
WARNING: Not overriding core package 'qttools'; use -f to force
WARNING: Not overriding core package 'rblibtorrent'; use -f to force
WARNING: Not overriding core package 'rclone'; use -f to force
WARNING: Not overriding core package 'rclone-ng'; use -f to force
WARNING: Not overriding core package 'rclone-webui-react'; use -f to force
WARNING: Not overriding core package 'redsocks2'; use -f to force
WARNING: Not overriding core package 'tcpping'; use -f to force
WARNING: Not overriding core package 'UnblockNeteaseMusic'; use -f to force
WARNING: Not overriding core package 'UnblockNeteaseMusic-Go'; use -f to force
WARNING: Not overriding core package 'uugamebooster'; use -f to force
WARNING: Not overriding core package 'verysync'; use -f to force
WARNING: Not overriding core package 'vlmcsd'; use -f to force
WARNING: Not overriding core package 'wxbase'; use -f to force
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed helloworld.
builder@Build-Server:~/lede_r4s$
```

GreatMichaelLee, 2022-02-14:
删除tmp和重新make menuconfig之后，还有四个

builder@Build-Server:~/lede_r4s$ ./scripts/feeds install -a
Installing all packages from feed packages.
WARNING: Not overriding core package 'dns2socks'; use -f to force
WARNING: Not overriding core package 'ipt2socks'; use -f to force
WARNING: Not overriding core package 'microsocks'; use -f to force
WARNING: Not overriding core package 'pdnsd-alt'; use -f to force

WYC-2020, 2022-02-14:
你是完全看不懂英语啊？别人这是是说 你如果要强制覆盖 请加入-f   库有重复的

GreatMichaelLee, 2022-02-14:
> 你是完全看不懂英语啊？别人这是是说 你如果要强制覆盖 请加入-f 库有重复的

我的经验是用-f force overriding的话，一般编译都会报错，如果上一版本OK的话，我一般都不用 -f.

closed 2022-02-14 by GreatMichaelLee

709397578, 2025-02-24:
我也遇到过，处理办法是，重新克隆一个库，从头开始，再做一遍就OK了