coolsnowwolf/lede #8876: 有用配置DNSMasq来获取IPTV（上海电信） B面IP地址的兄弟吗？

mrclock, 2022-02-14:
X86软路由，1月15日后的版本就开始有问题了。15日前一个版本也出过问题。去年11月的时候也有类似问题，后来更新代码后就好了。
周末我尝试了用默认配置，只配置了PPPoE、VLan85和DNSmasq，也是不行。同样配置1月15日的固件是可以用的。

所以想问问有没有同样情况的同学？

WYC-2020, 2022-02-14:
你这是让别人猜什么问题吗，说有问题又不说清除具体什么问题

mrclock, 2022-02-14:
> 你这是让别人猜什么问题吗，说有问题又不说清除具体什么问题

不好意思描述的不太清楚。可能只有用这项配置的能意会。

就是1/15后的版本，在DNSMasq里面加了手工加了指定配置后，IPTV B面的内网IP地址还是获取不到。

这项配置是用来，欺骗机顶盒用了电信自己的设备，这样才能获取到B面的内网IP，否则只能获取到A面的IP。
光有A面地址，IPTV是不能使用的。

WYC-2020, 2022-02-14:
上海光猫桥接+iptv 这个没玩过，看有人玩过没

lqzhj, 2022-02-14:
> X86软路由，1月15日后的版本就开始有问题了。15日前一个版本也出过问题。去年11月的时候也有类似问题，后来更新代码后就好了。 周末我尝试了用默认配置，只配置了PPPoE、VLan85和DNSmasq，也是不行。同样配置1月15日的固件是可以用的。
> 
> 所以想问问有没有同样情况的同学？

我也是默认配置好好的跟新后不工作了。

xiaoyouke01, 2022-02-14:
请教上海电信光猫+x86软路由是怎么配置iptv的，有教程吗，发下邮箱xiaoyouke@163.com，谢谢

WYC-2020, 2022-02-14:
@xiaoyouke01 很多文章啊，ytb还有视频来的，去看吧

mrclock, 2022-02-15:
昨天看到有一个DNSMASQ.init的更新，重新编译了下，故障依旧。日志里抠了一点出来

`Tue Feb 15 09:23:47 2022 daemon.info : 12[KNL] flags changed for fe80::2f1:f5ff:fe14:2af9 on br-VLAN_85
Tue Feb 15 09:23:47 2022 daemon.info : 15[KNL] flags changed for fe80::2f1:f5ff:fe14:2afa on br-lan
Tue Feb 15 09:23:48 2022 daemon.err modprobe: failed to find a module named aufs
Tue Feb 15 09:23:48 2022 daemon.err dockerd[3269]: time="2022-02-15T01:23:48.115804283Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Tue Feb 15 09:23:48 2022 daemon.err dockerd[3269]: time="2022-02-15T01:23:48.116638304Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Tue Feb 15 09:23:48 2022 daemon.info : 12[KNL] flags changed for fd8f:ddc2:9155::1 on br-lan
Tue Feb 15 09:23:48 2022 daemon.err dockerd[3269]: time="2022-02-15T01:23:48.282952458Z" level=error msg="failed to mount overlay: invalid argument" storage-driver=overlay2
Tue Feb 15 09:23:48 2022 daemon.err dockerd[3269]: time="2022-02-15T01:23:48.283202211Z" level=error msg="exec: \"fuse-overlayfs\": executable file not found in $PATH" storage-driver=fuse-overlayfs
Tue Feb 15 09:23:48 2022 kern.err kernel: [   47.261433] overlayfs: filesystem on '/opt/check-overlayfs-support229520189/upper' not supported as upperdir
Tue Feb 15 09:23:48 2022 daemon.err modprobe: failed to find a module named aufs
Tue Feb 15 09:23:48 2022 daemon.err dockerd[3269]: time="2022-02-15T01:23:48.290303930Z" level=error msg="AUFS was not found in /proc/filesystems" storage-driver=aufs
Tue Feb 15 09:23:48 2022 daemon.notice procd: /etc/rc.d/S90turboacc: DNSMASQ change
Tue Feb 15 09:23:48 2022 daemon.err dockerd[3269]: time="2022-02-15T01:23:48.301566493Z" level=error msg="failed to mount overlay: invalid argument" storage-driver=overlay
Tue Feb 15 09:23:48 2022 daemon.err dockerd[3269]: time="2022-02-15T01:23:48.301601330Z" level=error msg="Failed to built-in GetDriver graph devicemapper /opt"
Tue Feb 15 09:23:48 2022 kern.err kernel: [   47.281650] overlayfs: filesystem on '/opt/check-overlayfs-support4228168474/upper' not supported as upperdir
Tue Feb 15 09:23:48 2022 daemon.err dockerd[3269]: time="2022-02-15T01:23:48.549892419Z" level=warning msg="Your kernel does not support cgroup blkio weight"
Tue Feb 15 09:23:48 2022 daemon.err dockerd[3269]: time="2022-02-15T01:23:48.549924258Z" level=warning msg="Your kernel does not support cgroup blkio weight_device"

--------------------------------------------------
Tue Feb 15 09:24:03 2022 daemon.info procd: Instance dnsmasq::cfg01411c pid 5722 not stopped on SIGTERM, sending SIGKILL instead
`

mrclock, 2022-02-15:
> 请教上海电信光猫+x86软路由是怎么配置iptv的，有教程吗，发下邮箱[xiaoyouke@163.com](mailto:xiaoyouke@163.com)，谢谢

上海电信的话，做了桥接，路由拨号。把光猫3或者4口(SDN网关随意口)连到路由器Wan口，给Wan口和你要接机顶盒的LAN口做VLAN，标签是85，X86的路由器就是 eth0.85 eth1.85，其中ethX.85中的X是数字，代表你哪个Lan口，其他的路由器的话，百度看下教程。另外一个重要步骤就是，SCP连到路由器，修改etc目录下的dnsmasq.conf，里面添加如下代买，保存，重启不出意外就能看了。否则换下dhcp-option-force=125 那一行，百度搜索下你对应机器的代码。按理我这个是全部适用的。

只是最近的固件不起效。要1月15日编译的版本可以。

#IPTV
dhcp-option-force=125,00:00:00:00:1a:02:06:48:47:57:2d:43:54:03:08:45:2d:31:34:30:57:2d:50:0b:02:00:55:0a:02:20:00:3c:1e:00:00:01:00:02:03:43:50:45:03:0e:45:38:20:45:50:4f:4e:20:52:4f:55:54:45:52:04:03:31:2e:30
dhcp-option=15
dhcp-option=28
dhcp-option=60,00:00:01:06:68:75:61:71:69:6E:02:0A:48:47:55:34:32:31:4E:20:76:33:03:0A:48:47:55:34:32:31:4E:20:76:33:04:10:32:30:30:2E:55:59:59:2E:30:2E:41:2E:30:2E:53:48:05:04:00:01:00:50

mrclock, 2022-02-15:
更新回1月15日的版本

Tue Feb 15 10:03:59 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Feb 15 10:03:59 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses

dnsmasq有启动迹象

mrclock, 2022-02-17:
2月16日的版本 编译后，二次重启路由就能连IPTV了。关闭issues。

closed 2022-02-17 by mrclock