coolsnowwolf/lede #8877: dnsmasq运行时报错

Cptmacmillan2022007, 2022-02-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86-64-efi

### (3) 详细日志

A：

dnsmasq[20034]: segfault at 7fdccdff9c80 ip 00007fdcce16f4d3 sp 00007ffee057e388 error 4 in libc.so[7fdcce16c000+49000]
[  321.731454] Code: b6 03 84 c0 75 f1 48 89 d8 5b c3 0f 1f 40 00 67 e8 c2 03 00 00 48 01 d8 5b c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 0f b6 07 <0f> b6 16 38 c2 75 25 84 d2 74 2a b8 01 00 00 00 eb 0b 0f 1f 00 48

WYC-2020, 2022-02-15:
这个我看l大昨天更新了一下，你在编译试下看看还会出异常不

Cptmacmillan2022007, 2022-02-15:
> 这个我看l大昨天更新了一下，你在编译试下看看还会出异常不

没有那个异常了，但是fullconenat出问题了

一打开防火墙fullconenat的开关wan口马上断网

关掉恢复正常

内核是5.10的

这个是怎么回事啊？

Cptmacmillan2022007, 2022-02-17:
> 

好的，我用你的源码在线编译了

Cptmacmillan2022007, 2022-02-17:
> @Cptmacmillan2022007 我没遇到过这样的问题，你可到我仓库去下载我编译好的试下看看 会不会这样，因为我没法复现

好的，我用你的源码在线编译下试试

WYC-2020, 2022-02-17:
> > @Cptmacmillan2022007 我没遇到过这样的问题，你可到我仓库去下载我编译好的试下看看 会不会这样，因为我没法复现
> 
> 好的，我用你的源码在线编译下试试
> 
> 另外能修下qtbase下载的问题吗
> 
> 编译在下载的时候出现了
> 
> ERROR: package/feeds/packages/qtbase failed to build. make package/download: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
> 
> 这会影响到编译吗？
> 
> 我没有选择qtbase包啊
en
你是全新clone的？你没选不会编译才对，你怕是哪个包依赖了吧，选择qtorrent?

Cptmacmillan2022007, 2022-02-17:
> @Cptmacmillan2022007 我没遇到过这样的问题，你可到我仓库去下载我编译好的试下看看 会不会这样，因为我没法复现
>
>
> 好的，我用你的源码在线编译下试试
> 另外能修下qtbase下载的问题吗
> 编译在下载的时候出现了
> ERROR: package/feeds/packages/qtbase failed to build. make package/download: build [...]

我再重新手搓一份config试下

WYC-2020, 2022-02-18:
@Cptmacmillan2022007  下载的时候 会有地址打印出来 你手工下载放到dl目录即可

Cptmacmillan2022007, 2022-02-18:
> @Cptmacmillan2022007 下载的时候 会有地址打印出来 你手工下载放到dl目录即可

我是在线编译的没有下载地址啊

Cptmacmillan2022007, 2022-02-22:
> @Cptmacmillan2022007 下载的时候 会有地址打印出来 你手工下载放到dl目录即可

你源码的umbim编译出错
umbim的详细信息捕捉不出来只有这些

2022-02-21T07:14:22.6388008Z  make[3] -C package/network/utils/umbim compile
2022-02-21T07:14:23.2898594Z     ERROR: package/network/utils/umbim failed to build.

closed 2022-02-28 by Cptmacmillan2022007