coolsnowwolf/lede #8879: 今日更新最新代码后R2S编译编译报错

joesdu, 2022-02-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：今日(2022-02-14 15:05:56)更新最新代码后R2S编译编译报错

### (2) 路由器型号和固件版本

A：Nano Pi R2S

### (3) 详细日志

A：
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency frpc for luci-app-frpc
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-frpc found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-frpc.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-frpc-zh-cn:
 *      frpc
 * opkg_install_cmd: Cannot install package luci-i18n-frpc-zh-cn.
 * pkg_hash_check_unresolved: cannot find dependency pdnsd-alt for luci-app-turboacc
 * pkg_hash_check_unresolved: cannot find dependency dnsforwarder for luci-app-turboacc
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-turboacc found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-turboacc.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-turboacc-zh-cn:
 *      pdnsd-alt
 *      dnsforwarder
 * opkg_install_cmd: Cannot install package luci-i18n-turboacc-zh-cn.
 * pkg_hash_check_unresolved: cannot find dependency UnblockNeteaseMusic-Go for luci-app-unblockmusic
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-unblockmusic found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-unblockmusic.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-unblockmusic-zh-cn:
 *      UnblockNeteaseMusic-Go
 * opkg_install_cmd: Cannot install package luci-i18n-unblockmusic-zh-cn.
 * pkg_hash_check_unresolved: cannot find dependency dns2socks for luci-app-ssr-plus
 * pkg_hash_check_unresolved: cannot find dependency microsocks for luci-app-ssr-plus
 * pkg_hash_check_unresolved: cannot find dependency pdnsd-alt for luci-app-ssr-plus
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-ssr-plus found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-ssr-plus.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-ssr-plus-zh-cn:
 *      dns2socks
 *      microsocks
 *      pdnsd-alt
 * opkg_install_cmd: Cannot install package luci-i18n-ssr-plus-zh-cn.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/joe/r2s'
make[1]: *** [package/Makefile:111: /home/joe/r2s/staging_dir/target-aarch64_generic_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/joe/r2s'
make: *** [/home/joe/r2s/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-02-14:
我编译x86 没任何问题，你肯定是没更新源

wkq2004, 2022-02-14:
> 

更新什么源？

Acris, 2022-02-14:
Try:
```
./scripts/feeds update -a
./scripts/feeds install -a
```

wkq2004, 2022-02-14:
都更新了，编译就是出错。

WYC-2020, 2022-02-14:
英文看不懂吧 哈哈！,没问题的只是没有覆盖而已
@mrclock

zhengjun131461, 2022-02-14:
怎么才能覆盖哦，好几个源，两个源内有同一个App，该如果处理，

XinSSS, 2022-02-14:
make clean  还有什么feeds clean 我都用了， 不好使

用make distclean

closed 2022-02-15 by AmadeusGhost