coolsnowwolf/lede #8880: 2月14号版本编译报错

wokilp, 2022-02-14:
touch /home/test/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.luci-i18n-easymesh-zh-cn_installed
echo "luci-i18n-easymesh-zh-cn" >> /home/test/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/luci-app-easymesh.default.install
make[3]: Leaving directory '/home/test/lede/package/lean/luci-app-easymesh'
time: package/lean/luci-app-easymesh/compile#0.11#0.04#0.13
fatal: unable to access 'https://github.com/llccd/netfilter-full-cone-nat.git/': Failed to connect to github.com port 443: Connection refused
make[3]: *** [Makefile:66: /home/test/lede/dl/fullconenat-2022-01-29-1dffa4c8.tar.xz] Error 128
make[3]: Leaving directory '/home/test/lede/package/lean/openwrt-fullconenat'
time: package/lean/openwrt-fullconenat/compile#0.26#0.06#23.38
    ERROR: package/lean/openwrt-fullconenat failed to build.
make[2]: *** [package/Makefile:116: package/lean/openwrt-fullconenat/compile] Error 1
make[2]: Leaving directory '/home/test/lede'
make[1]: *** [package/Makefile:110: /home/test/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/test/lede'
make: *** [/home/test/lede/include/toplevel.mk:230: world] Error 2

wokilp, 2022-02-14:
make[2] -C package/utils/busybox download
 make[2] -C package/utils/jsonfilter download
 make[2] -C package/utils/lua download
 make[2] -C package/utils/mtd-utils download
 make[2] -C package/utils/util-linux download
    ERROR: package/lean/openwrt-fullconenat failed to build.
make package/download: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
 make[1] target/download
 make[2] -C target/linux download

libinanshu, 2022-02-14:
依次运行
cd lede
git pull
./scripts/feeds update -a
./scripts/feeds install -a
make clean
rm -rf ./tmp
rm -rf .config

AmadeusGhost, 2022-02-14:
你网络问题。。。

closed 2022-02-14 by AmadeusGhost