coolsnowwolf/lede #8883: 建议添加mac80211驱动修复

whoshoe, 2022-02-15:
在使用WRT32X时经常会遇到无线网络断开的问题，在openwrt论坛看到这个https://forum.openwrt.org/t/users-needed-to-test-wi-fi-stability-on-linksys-wrt3200acm-wrt32x-on-openwrt-21-02/101700帖子，该问题已经在openwrt 21.02 master和snapshot里面修复了。

这是一个regression的问题，修复的邮件可以参考如下：
https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=blob_plain;f=package/kernel/mac80211/patches/subsys/305-mac80211-fix-regression-in-SSN-handling-of-addba-tx.patch;h=dc8afb9186f25029312f85bca9b994fc3a604a66;hb=8e001a178219630a0c5254461e5d343f5f147c8e

由于此commit进linux stable还需要漫长的时间，希望能被添加到lede的代码库。

Beginner-Go, 2022-02-15:
这里的mac80211和官方差异太多，直接同步官方容易炸

WYC-2020, 2022-02-15:
@whoshoe  看了一下代码差不多，我这边加入了
https://github.com/WYC-2020/lede/commit/38ce186eb541a8bd464ff2388ad8b42edbb789ef

Beginner-Go, 2022-02-15:
> @whoshoe 看了一下代码差不多，我这边加入了 [WYC-2020/lede@38ce186](https://github.com/WYC-2020/lede/commit/38ce186eb541a8bd464ff2388ad8b42edbb789ef)

以前尝试同步过官方 容易炸

WYC-2020, 2022-02-15:
@Beginner-Go 这个我看源码位置都一样，所以就更新了，这个修复应该是针对所有用这个驱动的型号吧，暂时先放这里吧，顺便给他们想测试的一个入口，哈哈

Beginner-Go, 2022-02-15:
测试没问题我就去merged

WYC-2020, 2022-02-15:
@whoshoe 大兄弟去测试一下吧，最好有用到这个驱动的 都去测试一把，理论应该不会出问题，就一个patch而已

whoshoe, 2022-02-16:
> @whoshoe 大兄弟去测试一下吧，最好有用到这个驱动的 都去测试一把，理论应该不会出问题，就一个patch而已

好的，我先用你的repo编译一个固件试用一周看看。

closed 2022-02-16 by Beginner-Go