coolsnowwolf/lede #8884: 编译5.15内核时，提示缺少libphy.ko，编译失败

ricksuzade-maker, 2022-02-15:
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译kmod-phylink时，提示缺少libphy.ko，造成编译失败

### (2) 路由器型号和固件版本

A：x86-64 ESXi 7 虚拟机

### (3) 详细日志

A：
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.rt2800-usb-firmware_installed
echo "rt2800-usb-firmware" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/rt61-pci-firmware/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.rt61-pci-firmware_installed
echo "rt61-pci-firmware" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/rt73-usb-firmware/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.rt73-usb-firmware_installed
echo "rt73-usb-firmware" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/amdgpu-firmware/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.amdgpu-firmware_installed
echo "amdgpu-firmware" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/radeon-firmware/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.radeon-firmware_installed
echo "radeon-firmware" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/ar3k-firmware/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.ar3k-firmware_installed
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/ath9k-htc-firmware/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.ath9k-htc-firmware_installed
echo "ath9k-htc-firmware" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/carl9170-firmware/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.carl9170-firmware_installed
echo "carl9170-firmware" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/wil6210-firmware/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.wil6210-firmware_installed
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/r8152-firmware/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.r8152-firmware_installed
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/r8169-firmware/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.r8169-firmware_installed
echo "r8169-firmware" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/rtl8188eu-firmware/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.rtl8188eu-firmware_installed
echo "rtl8188eu-firmware" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/rtl8192ce-firmware/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.rtl8192ce-firmware_installed
echo "rtl8192ce-firmware" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
[... cut ...]
time: package/kernel/linux/compile#2.78#0.29#3.06
    ERROR: package/kernel/linux failed to build.
package/Makefile:114: recipe for target 'package/kernel/linux/compile' failed
make[1]: *** [package/kernel/linux/compile] Error 1
make[1]: Leaving directory '/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt'
/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/include/toplevel.mk:228: recipe for target 'package/compile' failed
make: *** [package/compile] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2022-02-15:
翻看类似issue 这个我说过

WYC-2020, 2022-02-15:
https://github.com/coolsnowwolf/lede/issues/8706

WYC-2020, 2022-02-15:
https://github.com/WYC-2020/lede/commit/6f4fc8e4aae1ba0c461762559b3842e7a91aeaae 
@ricksuzade-maker

ricksuzade-maker, 2022-02-15:
感谢，我去试试

closed 2022-02-16 by Beginner-Go