coolsnowwolf/lede #8886: 5.15 源码编译提示vfio_mdev.ko' is missing.

vipboddy, 2022-02-15:
make[2]: Entering directory '/mnt/sdb1/test/lede/package/kernel/linux'
ERROR: module '/mnt/sdb1/test/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.23/drivers/vfio/mdev/vfio_mdev.ko' is missing.
make[2]: *** [modules/virt.mk:99: /mnt/sdb1/test/lede/bin/targets/x86/64/packages/kmod-vfio-mdev_5.15.23-1_x86_64.ipk] Error 1
make[2]: Leaving directory '/mnt/sdb1/test/lede/package/kernel/linux'
time: package/kernel/linux/compile#2.43#0.22#2.64
    ERROR: package/kernel/linux failed to build.
make[1]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[1]: Leaving directory '/mnt/sdb1/test/lede'
make: *** [/mnt/sdb1/test/lede/include/toplevel.mk:230: package/compile] Error 2

WYC-2020, 2022-02-16:
下午给你看下
@vipboddy

WYC-2020, 2022-02-16:
https://github.com/WYC-2020/lede/commit/6a88351e71445d40af23dab8077f44d583a7f122
@vipboddy

closed 2022-02-16 by Beginner-Go