coolsnowwolf/lede #8887: R4S 编译报错 btrfs-progs错误

ZZMHA227, 2022-02-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
R4S 编译时， ERROR: package/feeds/packages/btrfs-progs failed to build.
Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：R4S 编译时， ERROR: package/feeds/packages/btrfs-progs failed to build.

### (2) 路由器型号和固件版本

A：R4S

### (3) 详细日志

A：checking for EXT4_EPOCH_MASK defined in ext2fs/ext2_fs.h... yes
checking linux/blkzoned.h usability... yes
checking linux/blkzoned.h presence... yes
checking for linux/blkzoned.h... yes
checking for struct blk_zone.capacity... no
checking for BLKGETZONESZ defined in linux/blkzoned.h... yes
configure: error: linux/blkzoned.h does not provide blk_zone.capacity
make[3]: *** [Makefile:100: /workdir/openwrt/build_dir/target-aarch64_generic_musl/btrfs-progs-v5.16.1/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/btrfs-progs'
time: package/feeds/packages/btrfs-progs/compile#1.28#0.64#1.55
    ERROR: package/feeds/packages/btrfs-progs failed to build.
make[2]: Leaving directory '/workdir/openwrt'
make[2]: *** [package/Makefile:116: package/feeds/packages/btrfs-progs/compile] Error 1
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

bleach1991, 2022-02-16:
一样

Mousehk, 2022-02-16:
同样问题

tpwonder, 2022-02-16:
一样的问题，昨天开始的

wekingchen, 2022-02-16:
一样的问题，N1盒子：

make[3]: *** [Makefile:100: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/btrfs-progs-v5.16.1/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/btrfs-progs'
time: package/feeds/packages/btrfs-progs/compile#1.46#0.95#1.95
    ERROR: package/feeds/packages/btrfs-progs failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/btrfs-progs/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

btguigui, 2022-02-16:
[ipq40xx]，也报错

WYC-2020, 2022-02-16:
https://github.com/kdave/btrfs-progs/commit/431dc7021c43e43658af436208182f8680e15fe2
@Mousehk 
@wekingchen 
@bleach1991 
@ZZMHA227 
@btguigui

tpwonder, 2022-02-16:
感觉问题处在btrfs的文件系统支持上，打算先不选择btrfs支持再compile试试看

WYC-2020, 2022-02-16:
算了我来做一个补丁吧

tpwonder, 2022-02-16:
> 算了我来做一个补丁吧

Thanks!

WYC-2020, 2022-02-16:
https://github.com/WYC-2020/packages/commit/74b200f20a1a6161b120be383b766e7b0ab1bdec
@Mousehk 
@wekingchen 
@bleach1991 
@ZZMHA227 
@btguigui

wekingchen, 2022-02-16:
@WYC-2020 谢谢，我试试

rm -rf ./feeds/packages/utils/btrfs-progs
svn co https://github.com/WYC-2020/packages/trunk/utils/btrfs-progs ./feeds/packages/utils/btrfs-progs

closed 2022-02-16 by Beginner-Go