coolsnowwolf/lede #8888: 极路由X C526A 问题

rshi0212, 2022-02-16:
极路由X C526A
最新编译
[image]
Netdata无法访问
[image]
DAWN无法启动
[image]
WiFi 2.4G 软起动路由器后丢失，拔电源硬启动WiFi 2.4G正常，两个WiFi 5G均正常。

rshi0212, 2022-02-16:
软启动后的日志

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.175 (rshi0212@Latitude-E6540) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4197-29c93682a)) #0 SMP Wed Feb 16 06:41:12 2022
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: HiWiFi C526A
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 130560
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 130560 pages, LIFO batch:31
[    0.000000] percpu: Embedded 15 pages/cpu s30156 r8192 d23092 u61440
[    0.000000] pcpu-alloc: s30156 r8192 d23092 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129408
[    0.000000] Kernel command line: ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait ubi.mtd=ubi root=/dev/ubiblock0_1
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 506740K/522240K available (5634K kernel code, 186K rwdata, 1480K rodata, 1024K init, 232K bss, 15500K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x2f0/0x49c with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
[    0.000007] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
[    0.000021] Switching to timer-based delay loop, resolution 20ns
[    0.000228] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.000246] pid_max: default: 32768 minimum: 301
[    0.000449] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000466] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001457] CPU: Testing write buffer coherency: ok
[    0.002495] Setting up static identity map for 0x80300000 - 0x80300060
[    0.002640] rcu: Hierarchical SRCU implementation.
[    0.003081] smp: Bringing up secondary CPUs ...
[    0.006172] smp: Brought up 1 node, 4 CPUs
[    0.006192] SMP: Total of 4 processors activated (384.00 BogoMIPS).
[    0.006200] CPU: All CPU(s) started in SVC mode.
[    0.010823] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.010976] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.011002] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.011204] pinctrl core: initialized pinctrl subsystem
[    0.012230] NET: Registered protocol family 16
[    0.012578] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.013829] cpuidle: using governor ladder
[    0.013876] cpuidle: using governor menu
[    0.014544] pstore: Registered ramoops as persistent store backend
[    0.014559] ramoops: using 0x80000@0x83f00000, ecc: 0
[    0.035243] cryptd: max_cpu_qlen set to 1000
[    0.038627] usbcore: registered new interface driver usbfs
[    0.038693] usbcore: registered new interface driver hub
[    0.038748] usbcore: registered new device driver usb
[    0.038796] pps_core: LinuxPPS API ver. 1 registered
[    0.038807] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.038832] PTP clock support registered
[    0.039560] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
[    0.040732] clocksource: Switched to clocksource arch_sys_counter
[    0.041519] thermal_sys: Registered thermal governor 'step_wise'
[    0.041805] NET: Registered protocol family 2
[    0.041987] IP idents hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.042762] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.042806] TCP established hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.042866] TCP bind hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.042956] TCP: Hash tables configured (established 4096 bind 4096)
[    0.043073] UDP hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.043110] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes, linear)
[... 931 lines cut ...]
Wed Feb 16 14:43:29 2022 daemon.info netdata[9066]: Initializing file /var/cache/netdata/cpu.cpu1/irq.db.
Wed Feb 16 14:43:29 2022 daemon.info netdata[9066]: Initializing file /var/cache/netdata/netdata.plugin_diskspace_dt/main.db.
Wed Feb 16 14:43:29 2022 daemon.info netdata[9066]: Initializing file /var/cache/netdata/cpu.cpu1/user.db.
[... cut ...]
Wed Feb 16 14:51:42 2022 daemon.notice hostapd: wlan1: STA-OPMODE-SMPS-MODE-CHANGED 28:c6:3f:f6:2a:11 off
Wed Feb 16 14:51:42 2022 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 28:c6:3f:f6:2a:11
Wed Feb 16 14:51:42 2022 daemon.info hostapd: wlan1: STA 28:c6:3f:f6:2a:11 RADIUS: starting accounting session DC83286C969ADFAC
Wed Feb 16 14:51:42 2022 daemon.info hostapd: wlan1: STA 28:c6:3f:f6:2a:11 WPA: pairwise key handshake completed (RSN)
Wed Feb 16 14:51:42 2022 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED 28:c6:3f:f6:2a:11

rshi0212, 2022-02-16:
硬启动后的日志

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.175 (rshi0212@Latitude-E6540) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4197-29c93682a)) #0 SMP Wed Feb 16 06:41:12 2022
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: HiWiFi C526A
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 130560
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 130560 pages, LIFO batch:31
[    0.000000] percpu: Embedded 15 pages/cpu s30156 r8192 d23092 u61440
[    0.000000] pcpu-alloc: s30156 r8192 d23092 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129408
[    0.000000] Kernel command line: ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait ubi.mtd=ubi root=/dev/ubiblock0_1
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 506740K/522240K available (5634K kernel code, 186K rwdata, 1480K rodata, 1024K init, 232K bss, 15500K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x2f0/0x49c with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
[    0.000007] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
[    0.000021] Switching to timer-based delay loop, resolution 20ns
[    0.000225] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.000241] pid_max: default: 32768 minimum: 301
[    0.000446] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000464] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001463] CPU: Testing write buffer coherency: ok
[    0.002490] Setting up static identity map for 0x80300000 - 0x80300060
[    0.002632] rcu: Hierarchical SRCU implementation.
[    0.003077] smp: Bringing up secondary CPUs ...
[    0.006158] smp: Brought up 1 node, 4 CPUs
[    0.006179] SMP: Total of 4 processors activated (384.00 BogoMIPS).
[    0.006187] CPU: All CPU(s) started in SVC mode.
[    0.010831] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.010984] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.011008] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.011214] pinctrl core: initialized pinctrl subsystem
[    0.012244] NET: Registered protocol family 16
[    0.012586] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.013837] cpuidle: using governor ladder
[    0.013886] cpuidle: using governor menu
[    0.014562] pstore: Registered ramoops as persistent store backend
[    0.014576] ramoops: using 0x80000@0x83f00000, ecc: 0
[    0.035254] cryptd: max_cpu_qlen set to 1000
[    0.038633] usbcore: registered new interface driver usbfs
[    0.038703] usbcore: registered new interface driver hub
[    0.038759] usbcore: registered new device driver usb
[    0.038810] pps_core: LinuxPPS API ver. 1 registered
[    0.038819] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.038845] PTP clock support registered
[    0.039570] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
[    0.040733] clocksource: Switched to clocksource arch_sys_counter
[    0.041514] thermal_sys: Registered thermal governor 'step_wise'
[    0.041800] NET: Registered protocol family 2
[    0.041979] IP idents hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.042745] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.042791] TCP established hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.042853] TCP bind hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.042941] TCP: Hash tables configured (established 4096 bind 4096)
[    0.043056] UDP hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.043094] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes, linear)
[... 958 lines cut ...]
Wed Feb 16 14:45:24 2022 daemon.info netdata[8825]: Initializing file /var/cache/netdata/cpu.cpu1/guest_nice.db.
Wed Feb 16 14:45:24 2022 daemon.info netdata[8825]: Initializing file /var/cache/netdata/netdata.plugin_diskspace/main.db.
Wed Feb 16 14:45:24 2022 daemon.info netdata[8825]: Initializing file /var/cache/netdata/cpu.cpu1/guest.db.
[... cut ...]
Wed Feb 16 14:45:24 2022 daemon.info netdata[8825]: Initializing file /var/cache/netdata/cpu.cpu2/irq.db.
Wed Feb 16 14:45:24 2022 daemon.info netdata[8825]: Initializing file /var/cache/netdata/cpu.cpu2/user.db.
Wed Feb 16 14:45:24 2022 daemon.info netdata[8825]: Initializing file /var/cache/netdata/cpu.cpu2/system.db.
Wed Feb 16 14:45:24 2022 daemon.info procd: Instance netdata::instance1 s in a crash loop 6 crashes, 1 seconds since last crash
Wed Feb 16 14:45:26 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of loopback (lo)

rshi0212, 2022-02-17:
> 其他主题会这样？

[image]

和主题有关系？NETDATA直接访问端口被拒绝。

WYC-2020, 2022-02-17:
我x86 没任何问题

rshi0212, 2022-02-17:
c526a hc5962 cr660x一样的配置编译出来都是相同问题。

WYC-2020, 2022-02-17:
你用我仓库的编译试下
@rshi0212

WYC-2020, 2022-02-17:
netdata 不是用的https吧，dawn我带没看，我编译一个x86 看下
@rshi0212

Klosw, 2022-04-07:
如果我没记错  软重启 2.4G起不来 是老毛病了

closed 2022-07-25 by aiamadeus