coolsnowwolf/lede #8890: 半个月没编译，今天5.15编译失败了

baodp, 2022-02-17:
Installing base-files (49-r4199-a30eae452) to root...
Installing libgcc1 (8.4.0-3) to root...
Installing libc (1.2.2-3) to root...
Installing libubox20210516 (2021-05-16-b14c4688-2) to root...
Installing libuci20130104 (2021-04-14-4b3db117-5) to root...
Installing libnl-tiny1 (2020-08-05-c291088f-2) to root...
Installing libubus20210603 (2021-06-03-a8cf6782-2) to root...
Installing libjson-c5 (0.15-2) to root...
Installing libblobmsg-json20210516 (2021-05-16-b14c4688-2) to root...
Installing ubusd (2021-06-03-a8cf6782-2) to root...
Installing ubus (2021-06-03-a8cf6782-2) to root...
Installing jshn (2021-05-16-b14c4688-2) to root...
Installing netifd (2021-06-04-50381d0a-1) to root...
Installing jsonfilter (2018-02-04-c7e938d6-1) to root...
Installing usign (2020-05-23-f1f65026-1) to root...
Installing openwrt-keyring (2021-02-20-49283916-2) to root...
Installing ubox (2020-10-25-9ef88681-2) to root...
Installing fstools (2021-08-14-2e3aca29-1) to root...
Installing fwtool (2019-11-12-8f7fe925-1) to root...
Installing grub2 (2.06-1) to root...
Installing grub2-efi (2.06-1) to root...
Installing kmod-button-hotplug (5.15.23-3) to root...
Installing kernel (5.15.23-1-87878138a81edce3f3cbbba5910f26d9) to root...
Installing kmod-input-core (5.15.23-1) to root...
Installing kmod-cryptodev (5.15.23+1.12-x86-1) to root...
Installing kmod-crypto-hash (5.15.23-1) to root...
Installing kmod-crypto-null (5.15.23-1) to root...
Installing kmod-crypto-aead (5.15.23-1) to root...
Installing kmod-crypto-manager (5.15.23-1) to root...
Installing kmod-crypto-authenc (5.15.23-1) to root...
Package kernel (5.15.23-1-87878138a81edce3f3cbbba5910f26d9) installed in root is up to date.
Installing kmod-scsi-core (5.15.23-1) to root...
Installing kmod-crypto-acompress (5.15.23-1) to root...
Package kmod-crypto-aead (5.15.23-1) installed in root is up to date.
Installing kmod-crypto-arc4 (5.15.23-1) to root...
Package kmod-crypto-authenc (5.15.23-1) installed in root is up to date.
Installing kmod-crypto-crc32 (5.15.23-1) to root...
Installing kmod-crypto-crc32c (5.15.23-1) to root...
Installing kmod-crypto-ecb (5.15.23-1) to root...
Package kmod-crypto-hash (5.15.23-1) installed in root is up to date.
Installing kmod-crypto-kpp (5.15.23-1) to root...
Installing kmod-crypto-lib-blake2s (5.15.23-1) to root...
Installing kmod-crypto-lib-chacha20 (5.15.23-1) to root...
Installing kmod-crypto-lib-chacha20poly1305 (5.15.23-1) to root...
Installing kmod-crypto-lib-poly1305 (5.15.23-1) to root...
Installing kmod-crypto-lib-curve25519 (5.15.23-1) to root...
Package kmod-crypto-lib-poly1305 (5.15.23-1) installed in root is up to date.
Package kmod-crypto-manager (5.15.23-1) installed in root is up to date.
Package kmod-crypto-null (5.15.23-1) installed in root is up to date.
Installing kmod-crypto-sha1 (5.15.23-1) to root...
Installing kmod-crypto-user (5.15.23-1) to root...
Installing kmod-fs-exfat (5.15.23-1) to root...
Installing kmod-nls-base (5.15.23-1) to root...
Installing kmod-fs-ext4 (5.15.23-1) to root...
Installing kmod-lib-crc16 (5.15.23-1) to root...
Installing kmod-fs-f2fs (5.15.23-1) to root...
Installing kmod-fs-ntfs3 (5.15.23-1) to root...
Installing kmod-fs-vfat (5.15.23-1) to root...
Installing kmod-nls-cp437 (5.15.23-1) to root...
Installing kmod-nls-iso8859-1 (5.15.23-1) to root...
Installing kmod-nls-utf8 (5.15.23-1) to root...
Installing kmod-hwmon-core (5.15.23-1) to root...
Installing kmod-i2c-core (5.15.23-1) to root...
Installing kmod-i2c-algo-bit (5.15.23-1) to root...
Installing kmod-hid (5.15.23-1) to root...
Installing kmod-input-evdev (5.15.23-1) to root...
Installing kmod-hid-generic (5.15.23-1) to root...
Package kmod-input-core (5.15.23-1) installed in root is up to date.
Package kmod-input-evdev (5.15.23-1) installed in root is up to date.
Installing kmod-lib-crc-ccitt (5.15.23-1) to root...
[... 622 lines cut ...]
Configuring kmod-igb.
Configuring kmod-igc.
Configuring kmod-bnx2.
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Configuring dnsmasq-full.
Configuring xray-core.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency dns2socks for luci-app-ssr-plus
 * pkg_hash_check_unresolved: cannot find dependency microsocks for luci-app-ssr-plus
 * pkg_hash_check_unresolved: cannot find dependency pdnsd-alt for luci-app-ssr-plus
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-ssr-plus found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-ssr-plus.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-ssr-plus-zh-cn:
 *      dns2socks
 *      microsocks
 *      pdnsd-alt
 * opkg_install_cmd: Cannot install package luci-i18n-ssr-plus-zh-cn.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/bdp/lede'
make[1]: *** [package/Makefile:111: /home/bdp/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/bdp/lede'
make: *** [/home/bdp/lede/include/toplevel.mk:230: world] Error 2
bdp@DESKTOP-2M8H23L:~/lede$

baodp, 2022-02-17:
运行 make clean   命令 然后重新配置 更新也不行。默认配置 加了IPV6,点选了5.15.

WYC-2020, 2022-02-17:
自己用的源不是lede的packages吧，这些依赖从lede仓库移动到了packages,用make clean 是不行的，用./scripts/feeds clean试下，如果还不行就万能大法 @baodp

baodp, 2022-02-17:
> 自己用的源不是lede的包吧，这些依赖从lede仓库移动到了包，用make clean是不行的，用。

必须是这个源啊。

Cptmacmillan2022007, 2022-02-17:
pkg hash 检查没有通过

lede最近的源码更新移动某些软件包到其他feeds仓库了

InvincibleDavid, 2022-02-17:
我的一直是/lede/package/lean 下的东西下载不成功，重新试了好多次；
全局kx，ubuntu20虚拟机里，明明编官方的都没有任何问题。

WYC-2020, 2022-02-17:
@baodp 是的必须这个源，你用其他的的保证必须有这些依赖库

WYC-2020, 2022-02-17:
> 

我没遇到过，下载不了

baodp, 2022-02-17:
解决了，删了自己电脑库里的东西。从新拉，单线程过编译了。。
 我直接WSL删了重装。。。。。。

closed 2022-02-17 by baodp