coolsnowwolf/lede #8891: 编译5.15内核出现问题，提示：kmod-mmc-spi 缺乏依赖  of_mmc_spi.ko

ricksuzade-maker, 2022-02-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败，提示缺乏依赖

### (2) 路由器型号和固件版本

A：ESXi 7.0 x86-64虚拟机

### (3) 详细日志

A：
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/iwl4965-firmware/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.iwl4965-firmware_installed
echo "iwl4965-firmware" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/iwlwifi-firmware-iwl100/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.iwlwifi-firmware-iwl100_installed
echo "iwlwifi-firmware-iwl100" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/iwlwifi-firmware-iwl1000/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.iwlwifi-firmware-iwl1000_installed
echo "iwlwifi-firmware-iwl1000" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/iwlwifi-firmware-iwl105/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.iwlwifi-firmware-iwl105_installed
echo "iwlwifi-firmware-iwl105" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/iwlwifi-firmware-iwl135/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.iwlwifi-firmware-iwl135_installed
echo "iwlwifi-firmware-iwl135" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/iwlwifi-firmware-iwl2000/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.iwlwifi-firmware-iwl2000_installed
echo "iwlwifi-firmware-iwl2000" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/iwlwifi-firmware-iwl2030/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.iwlwifi-firmware-iwl2030_installed
echo "iwlwifi-firmware-iwl2030" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/iwlwifi-firmware-iwl3160/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.iwlwifi-firmware-iwl3160_installed
echo "iwlwifi-firmware-iwl3160" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/iwlwifi-firmware-iwl3168/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.iwlwifi-firmware-iwl3168_installed
echo "iwlwifi-firmware-iwl3168" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/iwlwifi-firmware-iwl5000/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.iwlwifi-firmware-iwl5000_installed
echo "iwlwifi-firmware-iwl5000" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/iwlwifi-firmware-iwl5150/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.iwlwifi-firmware-iwl5150_installed
echo "iwlwifi-firmware-iwl5150" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/linux-firmware-20211216/.pkgdir/iwlwifi-firmware-iwl6000g2/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/ [... cut ...]
time: package/kernel/linux/compile#2.91#0.31#3.21
    ERROR: package/kernel/linux failed to build.
package/Makefile:114: recipe for target 'package/kernel/linux/compile' failed
make[1]: *** [package/kernel/linux/compile] Error 1
make[1]: Leaving directory '/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt'
/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/include/toplevel.mk:228: recipe for target 'package/compile' failed
make: *** [package/compile] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2022-02-17:
@ricksuzade-maker  你倒是说清楚 具体哪个内核版本啊

ricksuzade-maker, 2022-02-17:
标题上写了，下面忘了，是5.15.23

WYC-2020, 2022-02-17:
等会我看下

WYC-2020, 2022-02-17:
https://github.com/WYC-2020/lede/commit/cdd0e10d3710b9a63445574b1c98a370313f97e8
@ricksuzade-maker

closed 2022-02-28 by AmadeusGhost